luxonis/depthai #1038: [BUG] Failed install (line 222) on Raspberry Pi 4

Maxelon, 2023-05-25:
Sorry, fairly new to GitHub, so this may already be reported. I have just tried installing DepthAI on my Raspberry Pi 4, running the latest OS (lite) version. I have used all the supplied defaults for installation, but I get the following:

Failing command: ['/home/ken/Luxonis/venv/bin/python3', '-Im', 'ensurepip', '--upgrade', '--default-pip']

Failed installing dependencies. Could be a bug in the installer or unsupported platform. Open a bug report over at https://github.com/luxonis/depthai - exited with status 1 at line 222

Maxelon, 2023-05-25:
Just installed vent, as per instructions in error. Deleted Luxonis folder and tried again. I now get:

  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... -bash: line 232: 12106 Killed                  pip install pyqt5

Failed installing dependencies. Could be a bug in the installer or unsupported platform. Open a bug report over at https://github.com/luxonis/depthai - exited with status 137 at line 228

themarpe, 2023-05-25:
@Maxelon do you mind testing the following:
https://github.com/luxonis/depthai-python/pull/829