luxonis/depthai #1056: [BUG] `usb_speed` instead of `usbSpeed` to set USB speed in depthai-sdk

maxsitt, 2023-06-17:
**Describe the bug**
When connected via USB2, the following warning is printed:
```
/home/pi/sdkvenv/lib/python3.9/site-packages/depthai_sdk-1.11.0-py3.9.egg/depthai_sdk/oak_camera.py:305:
UsbWarning: Device connected in USB2 mode! This might cause some issues.
In such case, please try using a (different) USB3 cable, or force USB2 mode 'with OakCamera(usbSpeed='usb2') as oak:'
warnings.warn("Device connected in USB2 mode! This might cause some issues. "
```

However USB speed can only be set with `with OakCamera(usb_speed="usb2") as oak:`

**Minimal Reproducible Example**
```
from depthai_sdk import OakCamera

with OakCamera(usbSpeed='usb2') as oak:
    color = oak.create_camera('color')

    oak.visualize(color, fps=True, scale=2/3)
    oak.start(blocking=True)
```

**Expected behavior**
Force USB2 mode when connecting to OAK device. Works only with `with OakCamera(usb_speed="usb2") as oak:`.

**Attach system log**
```
{
    "architecture": "32bit ELF",
    "machine": "armv7l",
    "platform": "Linux-6.1.21-v7+-armv7l-with-glibc2.31",
    "processor": "",
    "python_build": "default Mar 12 2021 04:06:34",
    "python_compiler": "GCC 10.2.1 20210110",
    "python_implementation": "CPython",
    "python_version": "3.9.2",
    "release": "6.1.21-v7+",
    "system": "Linux",
    "version": "#1642 SMP Mon Apr  3 17:20:52 BST 2023",
    "win32_ver": "",
    "packages": [
        "certifi==2023.5.7",
        "charset-normalizer==3.1.0",
        "depthai==2.22.0.0",
        "depthai-pipeline-graph==0.0.5",
        "depthai-sdk==1.11.0",
        "idna==3.4",
        "marshmallow==3.19.0",
        "numpy==1.24.2",
        "opencv-contrib-python==4.5.5.62",
        "packaging==23.1",
        "pip==20.3.4",
        "pkg-resources==0.0.0",
        "PyTurboJPEG==1.7.1",
        "Qt.py==1.3.8",
        "requests==2.31.0",
        "sentry-sdk==1.25.1",
        "setuptools==44.1.1",
        "types-PySide2==5.15.2.1.5",
        "urllib3==2.0.3",
        "xmltodict==0.13.0"
    ],
    "usb": [
        "NoLib"
    ],
    "uname": [
        "Linux raspberrypi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l"
    ]
```

maxsitt, 2023-06-21:
In the [SDK docs](https://docs.luxonis.com/projects/sdk/en/latest/api_reference/#depthai_sdk.OakCamera.__init__) everything is correct, only the warning message that gets printed to console when connected via USB2 is misleading:

```
/home/pi/sdkvenv/lib/python3.9/site-packages/depthai_sdk/oak_camera.py:305:
UsbWarning: Device connected in USB2 mode! This might cause some issues.
In such case, please try using a (different) USB3 cable, or force USB2 mode 'with OakCamera(usbSpeed='usb2') as oak:'
warnings.warn("Device connected in USB2 mode! This might cause some issues. "
```

Erol444, 2023-06-21:
Thanks for the report! I just fixed it:)

closed 2023-06-21 by maxsitt