luxonis/depthai #1080: [Feature-Request] NixOS support would be ideal. :rocket:

karrson, 2023-07-11:
### Start with the `why`:
NixOS is an old yet highly powerful Linux distro, and a lot more people are starting to use it after realizing it's power.
So many packages are available on NixOS in a way that just works.
This is because of the way NixOS separates it's packages.
The package manager is declarative, meaning you simply modify a single config to decide what packages will be on the system.
Considering that this camera is for robots and probably other products as well, I could see people wanting to use NixOS, 
just to have control and the ability to very quickly get things working in the prototyping stage and beyond.
Overall, supporting NixOS would be a very good idea, and I don't think you should overlook this.

### Move to the `what`:
I would like DepthAI to be installable via the Nix package manager: https://search.nixos.org/packages?channel=23.05&from=0&size=50&sort=relevance&type=packages&query=depthai

### Move to the `how`:
You would unfortunately have to spend some time learning about NixOS if you haven't done so already.
While learning new distros can be time-consuming and annoying, 
it would be a great selling point for the camera to support this.
While I get that this may or may not be the highest priority for you personally, 
to me and a lot of other people this is extremely important 
and it will be more and more important as NixOS rises in popularity.

themarpe, 2023-07-13:
@karrson how do you intend to work with depthai-core library that is installable with Nix package manager?
Would building & installing locally be the main negative drawback?

karrson, 2023-07-14:
Hey! Sorry, just saw your comment.
I plan to work both in python 3.7 and through C++ by linking to it myself, depending on the scenario.