luxonis/depthai #1081: [BUG] {Failed to calibrate devices with only color cameras}

lincolnxlw, 2023-07-13:
**Describe the bug**
`calibrate.py` script failed to calibrate boards with only color cameras (OAK-1, OAK-1-Lite) or any custom boards with only rgb cameras defined.

**Minimal Reproducible Example**
Use the [board config](https://github.com/luxonis/depthai-boards/blob/main/boards/OAK-1-LITE.json) for OAK-1-Lite and run the calibration script by `python3 calibrate.py -s 2.5 -db -brd /config/OAK-1-LITE.json.json -ih`

**Expected behavior**
Since only color camera available in the device, calibration will only run the rgb calibration for the color camera.

**Screenshots**
Calibration script expecting `stereo_config` and get the following log
```
File "/depthai/calibrate.py", line 699, in capture_images_sync

leftStereo = self.board_config['cameras'][self.board_config['stereo_config']['left_cam']]['name']

KeyError: 'stereo_config'
```

MaticTonin, 2023-07-13:
Hi.
Yes, I see that the program always assumes that board_config includes stereo calibration even if board does not incluce one.
You can in that case just add before that line

` if 'stereo_config' in self.board_config['cameras']:
                            leftStereo =  self.board_config['cameras'][self.board_config['stereo_config']['left_cam']]['name']
                            rightStereo = self.board_config['cameras'][self.board_config['stereo_config']['right_cam']]['name']
                            print(f'Left Camera of stereo is {leftStereo} and right Camera of stereo is {rightStereo}')
`

themarpe, 2023-07-14:
@MaticTonin please open a PR with this changes such that our calibration procedure gets improved overall

MaticTonin, 2023-07-20:
#1083

themarpe, 2023-07-28:
@lincolnxlw does that resolve your issue? Feel free to reopen otherwise

closed 2023-07-28 by themarpe