luxonis/depthai #1084: [BUG] install_depthai.sh - 'libtbb2' replaced

negative-video, 2023-07-21:
Running the installation script on Ubuntu 23.04 using `sudo wget -qO- https://docs.luxonis.com/install_depthai.sh | bash` returns the following output before terminating:

> Package libtbb2 is not available, but is referred to by another package.
> This may mean that the package is missing, has been obsoleted, or
> is only available from another source
> However the following packages replace it:
>   libtbbmalloc2
> 
>E: Package 'libtbb2' has no installation candidate
> 
> Failed installing dependencies. Could be a bug in the installer or unsupported platform. Open a bug report over at https://github.com/luxonis/depthai - exited with status 100 at line 116 
>
>
> Failed installing dependencies. Could be a bug in the installer or unsupported platform. Open a bug report over at https://github.com/luxonis/depthai - exited with status 100 at line 192 

As the output implies, it sounds like the script at https://docs.luxonis.com/install_dependencies.sh should be modified within `readonly ubuntu_pkgs=(` on line 25 such that `libtbb2` is replaced with `libtbbmalloc2`.

brmarkus, 2023-07-21:
Seen this often with non-LTS releases... would you have a change to check with e.g. Ubuntu 22.04 (LTS) (or do a quick check within a Ubuntu-LTS docker container)?

negative-video, 2023-07-21:
I spun up a 22.04.2 Desktop AMD64 virtual machine and the DepthAI installation process completed successfully.

Perhaps `libtbb2` should be moved from `ubuntu_pkgs` into `ubuntu_pkgs_pre22_04`, while `libtbbmalloc2` should be inside `ubuntu_pkgs_post22_04` within the install_dependencies.sh script ?