luxonis/depthai #1085: [BUG] `Couldn't read data from stream: 'color'`

stuartcrobinson, 2023-07-24:
--> i'm using the usb-c cable that goes with my macbook charger <-- not sure if that could be the issue or not

after some installation hiccups, i got the demo to load but it seemed to hang on `Starting demo...` for a while until i finally got this error:

```
  File "/Users/stuartrobinson/Luxonis/depthai/depthai_demo.py", line 656, in run
    self.instance.run_all(self.conf)
  File "/Users/stuartrobinson/Luxonis/depthai/depthai_demo.py", line 133, in run_all
    self.run()
  File "/Users/stuartrobinson/Luxonis/depthai/depthai_demo.py", line 317, in run
    self.loop()
  File "/Users/stuartrobinson/Luxonis/depthai/depthai_demo.py", line 373, in loop
    self._pv.prepareFrames(callback=self.onNewFrame)
  File "/Users/stuartrobinson/.local/lib/python3.10/site-packages/depthai_sdk/managers/preview_manager.py", line 148, in prepareFrames
    packet = queue.tryGet()
RuntimeError: Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'color' (X_LINK_ERROR)'
```

[image]

[image]

macbook pro
2017 intel core i7
Monterey 12.1

below is the output from my installation experience.  i got the error about `git`, shown below, so then i ran `rm '/usr/local/bin/git'` like it suggested and it seemed to work.  some dependencies were missing so i ran `python3 install_requirements.py`, and then i ran `pip install opencv-python` to appease some other errors.  

running `python3 install_requirements.py` now comes back clean

so now it looks like all the requirements and dependencies are there but the demo never actually starts. and then fails out with `Couldn't read data from stream: 'color'` 

```
~ $ bash -c "$(curl -fL https://docs.luxonis.com/install_depthai.sh)"

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  7965  100  7965    0     0   7126      0  0:00:01  0:00:01 --:--:-- 5113k

ENTER absolute installation path for depthai or leave empty and default path: $HOME will be used.

Using default installation path: /Users/stuartrobinson/Luxonis
Directory: /Users/stuartrobinson/Luxonis is OK

Python version: Python 3.10.0 found.
If you want to use it for installation, press ANY key, otherwise input path to python binary.
Press any key to continue

_____________________________
Calling macOS_installer.sh
_____________________________
Running macOS installer.
Installing global dependencies.
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  5671  100  5671    0     0   4597      0  0:00:01  0:00:01 --:--:--  4597
During Homebrew install, certain commands need 'sudo'. Requesting access...
Password:

 >> Installing Homebrew from https://raw.githubusercontent.com/Homebrew/install/master/install.sh
Warning: Running in non-interactive mode because `$CI` is set.
==> Checking for `sudo` access (which may request your password)...
==> This script will install:
/usr/local/bin/brew
/usr/local/share/doc/homebrew
/usr/local/share/man/man1/brew.1
/usr/local/share/zsh/site-functions/_brew
/usr/local/etc/bash_completion.d/brew
[... 243 lines cut ...]
/usr/local/share/man/man7/giteveryday.7 -> /usr/local/git/share/man/man7/giteveryday.7
/usr/local/share/man/man7/gitglossary.7 -> /usr/local/git/share/man/man7/gitglossary.7
/usr/local/share/man/man7/gitnamespaces.7 -> /usr/local/git/share/man/man7/gitnamespaces.7
/usr/local/share/man/man7/gitremote-helpers.7 -> /usr/local/git/share/man/man7/gitremote-helpers.7
/usr/local/share/man/man7/gitrevisions.7 -> /usr/local/git/share/man/man7/gitrevisions.7
/usr/local/share/man/man7/gitsubmodules.7 -> /usr/local/git/share/man/man7/gitsubmodules.7
/usr/local/share/man/man7/gittutorial-2.7 -> /usr/local/git/share/man/man7/gittutorial-2.7
/usr/local/share/man/man7/gittutorial.7 -> /usr/local/git/share/man/man7/gittutorial.7
/usr/local/share/man/man7/gitworkflows.7 -> /usr/local/git/share/man/man7/gitworkflows.7
==> Caveats
The Tcl/Tk GUIs (e.g. gitk, git-gui) are now in the `git-gui` formula.
Subversion interoperability (git-svn) is now in the `git-svn` formula.

zsh completions and functions have been installed to:
  /usr/local/share/zsh/site-functions
==> Summary
🍺  /usr/local/Cellar/git/2.41.0_2: 1,633 files, 48.8MB
==> Running `brew cleanup git`...
Disable this behaviour by setting HOMEBREW_NO_INSTALL_CLEANUP.
Hide these hints with HOMEBREW_NO_ENV_HINTS (see `man brew`).

Failed installing dependencies. Could be a bug in the installer or unsupported platform. Open a bug report over at https://github.com/luxonis/depthai - exited with status 1 at line 164

Failed installing dependencies. Could be a bug in the installer or unsupported platform. Open a bug report over at https://github.com/luxonis/depthai - exited with status 1 at line 239
```

themarpe, 2023-07-25:
@PetrNovota the git installation could be done conditionally if no `git` exists.

@stuartcrobinson WRT the error, I'd suggest replacing USB cable & or trying USB2 mode, to see if issue persists

stuartcrobinson, 2023-07-25:
thanks for the quick response! 

ok i tried `python3 depthai_demo.py -usbs usb2`, which i found here https://docs.luxonis.com/en/latest/pages/troubleshooting/#forcing-usb2-communication,  but i seem to be getting the same error:

```
~/L/depthai $ python3 depthai_demo.py -usbs usb2

Using depthai module from:  /Users/stuartrobinson/.local/lib/python3.10/site-packages/depthai.cpython-310-darwin.so
Depthai version installed:  2.21.2.0
qt.qpa.fonts: Populating font family aliases took 205 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost.
Setting up demo...
Available devices:
[0] 184430109175610E00 [X_LINK_UNBOOTED]
USB Connection speed: UsbSpeed.HIGH
Enabling low-bandwidth mode due to low USB speed... (speed: UsbSpeed.HIGH)
Creating MJPEG link for MonoCamera node and left xlink stream...
Creating MJPEG link for MonoCamera node and right xlink stream...
Creating MJPEG link for ColorCamera node and color xlink stream...
/Users/stuartrobinson/.local/lib/python3.10/site-packages/depthai_sdk/previews.py:186: RuntimeWarning: invalid value encountered in cast
  dispFrame = (dispFrame * 255. / dispIntegerLevels).astype(np.uint8)
Stopping demo...
=== TOTAL FPS ===
[right]: 17.8
[left]: 12.7
[rectifiedLeft]: 11.1
[rectifiedRight]: 5.9
[depthRaw]: 6.2
[depth]: 6.2
  File "/Users/stuartrobinson/Luxonis/depthai/depthai_demo.py", line 656, in run
    self.instance.run_all(self.conf)
  File "/Users/stuartrobinson/Luxonis/depthai/depthai_demo.py", line 133, in run_all
    self.run()
  File "/Users/stuartrobinson/Luxonis/depthai/depthai_demo.py", line 317, in run
    self.loop()
  File "/Users/stuartrobinson/Luxonis/depthai/depthai_demo.py", line 373, in loop
    self._pv.prepareFrames(callback=self.onNewFrame)
  File "/Users/stuartrobinson/.local/lib/python3.10/site-packages/depthai_sdk/managers/preview_manager.py", line 148, in prepareFrames
    packet = queue.tryGet()
RuntimeError: Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'color' (X_LINK_ERROR)'
```

stuartcrobinson, 2023-07-25:
i'm assuming this means that trying a different usb-c cable (one that's definitely usb3) would **_not_** help?

themarpe, 2023-07-25:
@stuartcrobinson Yeah, unlikely but still do give it a try. 
This looks more like a power issue - try using the Y-Adapter or connect to a powered USB hub instead

stuartcrobinson, 2023-07-26:
@themarpe btw i just checked and discovered that the `depth` and rectified left and right views are working fine.  (so it's just like on my depthai rasp pi image now). i ordered a y-adapter yesterday so i'll try that out soon

but it's still just a still image even tho it looks like it should being showing 30 fps video for everything

stuartcrobinson, 2023-07-26:
@Erol444 responding to your suggestion https://github.com/luxonis/depthai/issues/1090#issuecomment-1651749441 to run `cam_test.py` in the rasp pi thread, i'm running it here too on my mac because they seem to be having the same behavior so far

note this error:

**`python3.10/site-packages/depthai_sdk/previews.py:186: RuntimeWarning: invalid value encountered in cast
  dispFrame = (dispFrame * 255. / dispIntegerLevels).astype(np.uint8)`**

**macbook pro:**
```
~/L/depthai $ python3 depthai_demo.py
Using depthai module from:  /Users/stuartrobinson/.local/lib/python3.10/site-packages/depthai.cpython-310-darwin.so
Depthai version installed:  2.21.2.0
qt.qpa.fonts: Populating font family aliases took 206 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost.
Setting up demo...
Available devices:
[0] 184430109175610E00 [X_LINK_UNBOOTED]
USB Connection speed: UsbSpeed.HIGH
Enabling low-bandwidth mode due to low USB speed... (speed: UsbSpeed.HIGH)
Creating MJPEG link for MonoCamera node and left xlink stream...
Creating MJPEG link for MonoCamera node and right xlink stream...
Creating MJPEG link for ColorCamera node and color xlink stream...
/Users/stuartrobinson/.local/lib/python3.10/site-packages/depthai_sdk/previews.py:186: RuntimeWarning: invalid value encountered in cast
  dispFrame = (dispFrame * 255. / dispIntegerLevels).astype(np.uint8)
^CStopping demo...
=== TOTAL FPS ===
[right]: 13.2
[left]: 17.6
[rectifiedLeft]: 6.3
[rectifiedRight]: 6.6
[depthRaw]: 6.8
[depth]: 6.8
```
```
~/r/d/utilities $ python3 cam_test.py -rs
Enabled cameras:
    rgb : color
   left : mono
  right : mono
   camd : color
DepthAI version: 2.21.2.0
DepthAI path: /Users/stuartrobinson/.local/lib/python3.10/site-packages/depthai.cpython-310-darwin.so
[184430109175610E00] [20.6] [0.938] [ColorCamera(9)] [error] Camera not detected on socket: 3
[184430109175610E00] [20.6] [0.939] [MonoCamera(7)] [error] OV7251 only supports THE_480_P/THE_400_P resolutions, defaulting to THE_480_P
[184430109175610E00] [20.6] [0.939] [MonoCamera(5)] [error] OV7251 only supports THE_480_P/THE_400_P resolutions, defaulting to THE_480_P
Connected cameras:
 -socket RGB   : IMX214 4208 x 3120 focus:auto  - COLOR
 -socket LEFT  : OV7251  640 x  480 focus:fixed - MONO
 -socket RIGHT : OV7251  640 x  480 focus:fixed - MONO
USB speed: HIGH
IR drivers: []
Cam:      rgb           left          right         camd     [host | capture timestamp]
FPS:   0.00|  0.00  29.75| 30.01  29.76| 30.01   0.00|  0.00
```

Erol444, 2023-07-26:
This indeed is a hardware issue. Color camera (IMX214) gets recognized and says it's connected, but it doesn't stream anything (FPS 0 at RGB). Please contact support@luxonis.com if such error occurs.

closed 2023-07-26 by Erol444