luxonis/depthai #1086: [BUG] Ubuntu, `install_requirements.py` cant download `depthai_calib.json`

stuartcrobinson, 2023-07-24:
trying to try out my new oak-d lite.  failed trying to install on my mac just now so now i've moved over to my `Ubuntu 20.04.5 LTS` laptop

`python3 install_requirements.py` cant seem to download this file:

https://artifacts.luxonis.com/artifactory/luxonis-depthai-data-local/misc/depthai_calib.json (error below)

and when i try to load that url directly my browser times out. 

and then when i try to run `python3 ColorCamera/rgb_preview.py`, i get a little output but nothing ever happens

```
stuart@hp:~/repos/depthai-python/examples$ python3 ColorCamera/rgb_preview.py
Connected cameras: [{socket: CAM_A, sensorName: IMX214, width: 4208, height: 3120, orientation: AUTO, supportedTypes: [COLOR], hasAutofocus: 1, name: color}, {socket: CAM_B, sensorName: OV7251, width: 640, height: 480, orientation: AUTO, supportedTypes: [MONO], hasAutofocus: 0, name: left}, {socket: CAM_C, sensorName: OV7251, width: 640, height: 480, orientation: AUTO, supportedTypes: [MONO], hasAutofocus: 0, name: right}]
Usb speed: SUPER
Device name: OAK-D-LITE
```

(nothing happens now.  "a small window video display" never appears.  (following https://docs.luxonis.com/projects/api/en/latest/install/#supported-platforms)

```
stuart@hp:~/repos/depthai-python/examples$ python3 install_requirements.py 
pip 23.2.1 from /home/stuart/.local/lib/python3.8/site-packages/pip (python 3.8)
Requirement already satisfied: pip in /home/stuart/.local/lib/python3.8/site-packages (23.2.1)
DEPRECATION: distro-info 0.23ubuntu1 has a non-standard version number. pip 23.3 will enforce this behaviour change. A possible replacement is to upgrade to a newer version of distro-info or contact the author to suggest that they release a version with a conforming version number. Discussion can be found at https://github.com/pypa/pip/issues/12063
DEPRECATION: gpg 1.13.1-unknown has a non-standard version number. pip 23.3 will enforce this behaviour change. A possible replacement is to upgrade to a newer version of gpg or contact the author to suggest that they release a version with a conforming version number. Discussion can be found at https://github.com/pypa/pip/issues/12063
DEPRECATION: python-debian 0.1.36ubuntu1 has a non-standard version number. pip 23.3 will enforce this behaviour change. A possible replacement is to upgrade to a newer version of python-debian or contact the author to suggest that they release a version with a conforming version number. Discussion can be found at https://github.com/pypa/pip/issues/12063
Requirement already satisfied: pyyaml in /home/stuart/.local/lib/python3.8/site-packages (6.0.1)
Requirement already satisfied: requests in /home/stuart/.local/lib/python3.8/site-packages (2.31.0)
Requirement already satisfied: numpy in /home/stuart/.local/lib/python3.8/site-packages (1.24.4)
Requirement already satisfied: opencv-contrib-python==4.5.5.62 in /home/stuart/.local/lib/python3.8/site-packages (4.5.5.62)
Requirement already satisfied: charset-normalizer<4,>=2 in /home/stuart/.local/lib/python3.8/site-packages (from requests) (3.2.0)
Requirement already satisfied: idna<4,>=2.5 in /usr/lib/python3/dist-packages (from requests) (2.8)
Requirement already satisfied: urllib3<3,>=1.21.1 in /usr/lib/python3/dist-packages (from requests) (1.25.8)
Requirement already satisfied: certifi>=2017.4.17 in /usr/lib/python3/dist-packages (from requests) (2019.11.28)
DEPRECATION: distro-info 0.23ubuntu1 has a non-standard version number. pip 23.3 will enforce this behaviour change. A possible replacement is to upgrade to a newer version of distro-info or contact the author to suggest that they release a version with a conforming version number. Discussion can be found at https://github.com/pypa/pip/issues/12063
DEPRECATION: gpg 1.13.1-unknown has a non-standard version number. pip 23.3 will enforce this behaviour change. A possible replacement is to upgrade to a newer version of gpg or contact the author to suggest that they release a version with a conforming version number. Discussion can be found at https://github.com/pypa/pip/issues/12063
DEPRECATION: python-debian 0.1.36ubuntu1 has a non-standard version number. pip 23.3 will enforce this behaviour change. A possible replacement is to upgrade to a newer version of python-debian or contact the author to suggest that they release a version with a conforming version number. Discussion can be found at https://github.com/pypa/pip/issues/12063
Collecting depthai
  Obtaining dependency information for depthai from https://files.pythonhosted.org/packages/a2/b2/7656a49d55f445501c3ebb3f8e10251f84afa74ee683705adc96b6e1cf4b/depthai-2.22.0.0-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata
  Using cached depthai-2.22.0.0-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (8.7 kB)
Using cached depthai-2.22.0.0-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (10.6 MB)
DEPRECATION: distro-info 0.23ubuntu1 has a non-standard version number. pip 23.3 will enforce this behaviour change. A possible replacement is to upgrade to a newer version of distro-info or contact the author to suggest that they release a version with a conforming version number. Discussion can be found at https://github.com/pypa/pip/issues/12063
DEPRECATION: gpg 1.13.1-unknown has a non-standard version number. pip 23.3 will enforce this behaviour change. A possible replacement is to upgrade to a newer version of gpg or contact the author to suggest that they release a version with a conforming version number. Discussion can be found at https://github.com/pypa/pip/issues/12063
DEPRECATION: python-debian 0.1.36ubuntu1 has a non-standard version number. pip 23.3 will enforce this behaviour change. A possible replacement is to upgrade to a newer version of python-debian or contact the author to suggest that they release a version with a conforming version number. Discussion can be found at https://github.com/pypa/pip/issues/12063
Installing collected packages: depthai
  Attempting uninstall: depthai
    Found existing installation: depthai 2.22.0.0
    Uninstalling depthai-2.22.0.0:
      Successfully uninstalled depthai-2.22.0.0
Successfully installed depthai-2.22.0.0
################|| Downloading calibration-sample ||################

========== Downloading /home/stuart/repos/depthai-python/examples/models/depthai_calib.json
Traceback (most recent call last):
  File "/home/stuart/repos/depthai-python/examples/downloader/downloader.py", line 82, in try_download
    chunk_iterable, continue_offset = start_download(offset=progress.size)
  File "/home/stuart/repos/depthai-python/examples/downloader/common.py", line 357, in start_download
    response.raise_for_status()
  File "/home/stuart/.local/lib/python3.8/site-packages/requests/models.py", line 1021, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 522 Server Error:  for url: https://artifacts.luxonis.com/artifactory/luxonis-depthai-data-local/misc/depthai_calib.json
########## Error: Download failed
```

themarpe, 2023-07-25:
@stuartcrobinson 

Do you mind trying again - seems like artifacts were down at that moment. Should work now

stuartcrobinson, 2023-07-25:
@themarpe 

ok yeah `install_requirements.py` appears to run successfully now, but `rgb_preview.py` still isn't doing anything.  it just hangs here forever, no other window pops up or anything:

```
stuart@hp:~/repos/depthai-python/examples/ColorCamera$ python3 rgb_preview.py 
Connected cameras: [{socket: CAM_A, sensorName: IMX214, width: 4208, height: 3120, orientation: AUTO, supportedTypes: [COLOR], hasAutofocus: 1, name: color}, {socket: CAM_B, sensorName: OV7251, width: 640, height: 480, orientation: AUTO, supportedTypes: [MONO], hasAutofocus: 0, name: left}, {socket: CAM_C, sensorName: OV7251, width: 640, height: 480, orientation: AUTO, supportedTypes: [MONO], hasAutofocus: 0, name: right}]
Usb speed: SUPER
Device name: OAK-D-LITE
```

themarpe, 2023-07-25:
CC: @jakaskerl

jakaskerl, 2023-07-26:
@stuartcrobinson 
Can you try running cam_test.py. I think you might be having issues with QT or OpenCV and cam_test.py should probably return a thread error.

closed 2023-07-26 by stuartcrobinson