luxonis/depthai #1087: [BUG] `Press any key to continue` is a trap

stuartcrobinson, 2023-07-24:
```
stuart@hp:~$ sudo wget -qO- https://docs.luxonis.com/install_depthai.sh | bash
[sudo] password for stuart: 

ENTER absolute installation path for depthai or leave empty and default path: $HOME will be used.

Using default installation path: /home/stuart/Luxonis
Directory: /home/stuart/Luxonis is OK

Python version: Python 3.8.10 found.
If you want to use it for installation, press ANY key, otherwise input path to python binary.
Press any key to continue
f
bash: line 90: f: command not found

Failed installing dependencies. Could be a bug in the installer or unsupported platform. Open a bug report over at https://github.com/luxonis/depthai - exited with status 127 at line 90 
```

themarpe, 2023-07-25:
Thanks for the report @stuartcrobinson - I assume ENTER has to be pressed instead? Does that work?
We'll modify the text if that's the case

stuartcrobinson, 2023-07-25:
ah, sorry. yes exactly. you're welcome and thank you!

themarpe, 2023-07-26:
Fixed in luxonis/depthai-python@236e7ec1b8ac2c7573d980f45efe90e9b3df812f

closed 2023-07-26 by themarpe