luxonis/depthai #1088: [BUG] raspberry pi: `Third party libraries failed to import: libhdf5_serial.so.103`

stuartcrobinson, 2023-07-25:
trying to set up my oak-d lite on a brand new raspberry pi 4

```
stuart@mypi:~/Luxonis/depthai $ python3 depthai_demo.py
Third party libraries failed to import: libhdf5_serial.so.103: cannot open shared object file: No such file or directory
Run "python3 install_requirements.py" to install dependencies
```

when i ran `python3 install_requirements.py`, everything looked good except for: `WARNING: Skipping opencv-python as it is not installed.`

any help would be super appreciated.  i opened tickets for my struggles to get this working on mac and a linux laptop but this is the only one that necessarily matters for me

full output:

```
stuart@mypi:~/Luxonis/depthai $ python3 install_requirements.py 
pip 23.2.1 from /home/stuart/.local/lib/python3.9/site-packages/pip (python 3.9)
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pip in /home/stuart/.local/lib/python3.9/site-packages (23.2.1)
WARNING: Skipping opencv-python as it is not installed.
Found existing installation: opencv-contrib-python 4.5.5.62
Uninstalling opencv-contrib-python-4.5.5.62:
  Successfully uninstalled opencv-contrib-python-4.5.5.62
Found existing installation: depthai 2.20.2.0
Uninstalling depthai-2.20.2.0:
  Successfully uninstalled depthai-2.20.2.0
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://www.piwheels.org/simple, https://artifacts.luxonis.com/artifactory/luxonis-depthai-data-local/wheels/, https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local/
Ignoring pyqt5: markers 'platform_machine != "armv6l" and platform_machine != "armv7l" and platform_machine != "aarch64" and platform_machine != "arm64"' don't match your environment
Requirement already satisfied: requests==2.26.0 in /home/stuart/.local/lib/python3.9/site-packages (from -r requirements.txt (line 1)) (2.26.0)
Requirement already satisfied: numpy>=1.21.4 in /home/stuart/.local/lib/python3.9/site-packages (from -r requirements.txt (line 3)) (1.25.1)
Collecting opencv-contrib-python==4.5.5.62 (from -r requirements.txt (line 4))
  Using cached https://www.piwheels.org/simple/opencv-contrib-python/opencv_contrib_python-4.5.5.62-cp39-cp39-linux_armv7l.whl (17.3 MB)
Requirement already satisfied: depthai-sdk==1.9.4 in /home/stuart/.local/lib/python3.9/site-packages (from -r requirements.txt (line 5)) (1.9.4)
Collecting depthai==2.20.2.0 (from -r requirements.txt (line 9))
  Using cached depthai-2.20.2.0-cp39-cp39-linux_armv7l.whl (11.6 MB)
Requirement already satisfied: Qt.py in /home/stuart/.local/lib/python3.9/site-packages (from -r requirements.txt (line 10)) (1.3.8)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in /usr/lib/python3/dist-packages (from requests==2.26.0->-r requirements.txt (line 1)) (1.26.5)
Requirement already satisfied: certifi>=2017.4.17 in /usr/lib/python3/dist-packages (from requests==2.26.0->-r requirements.txt (line 1)) (2020.6.20)
Requirement already satisfied: charset-normalizer~=2.0.0 in /home/stuart/.local/lib/python3.9/site-packages (from requests==2.26.0->-r requirements.txt (line 1)) (2.0.12)
Requirement already satisfied: idna<4,>=2.5 in /usr/lib/python3/dist-packages (from requests==2.26.0->-r requirements.txt (line 1)) (2.10)
Requirement already satisfied: blobconverter>=1.2.8 in /home/stuart/.local/lib/python3.9/site-packages (from depthai-sdk==1.9.4->-r requirements.txt (line 5)) (1.4.2)
Requirement already satisfied: pytube>=12.1.0 in /home/stuart/.local/lib/python3.9/site-packages (from depthai-sdk==1.9.4->-r requirements.txt (line 5)) (15.0.0)
Requirement already satisfied: PyTurboJPEG==1.6.4 in /home/stuart/.local/lib/python3.9/site-packages (from depthai-sdk==1.9.4->-r requirements.txt (line 5)) (1.6.4)
Requirement already satisfied: marshmallow==3.17.0 in /home/stuart/.local/lib/python3.9/site-packages (from depthai-sdk==1.9.4->-r requirements.txt (line 5)) (3.17.0)
Requirement already satisfied: distinctipy in /home/stuart/.local/lib/python3.9/site-packages (from depthai-sdk==1.9.4->-r requirements.txt (line 5)) (1.2.2)
Requirement already satisfied: xmltodict in /home/stuart/.local/lib/python3.9/site-packages (from depthai-sdk==1.9.4->-r requirements.txt (line 5)) (0.13.0)
Requirement already satisfied: packaging>=17.0 in /home/stuart/.local/lib/python3.9/site-packages (from marshmallow==3.17.0->depthai-sdk==1.9.4->-r requirements.txt (line 5)) (23.1)
Requirement already satisfied: types-PySide2 in /home/stuart/.local/lib/python3.9/site-packages (from Qt.py->-r requirements.txt (line 10)) (5.15.2.1.5)
Requirement already satisfied: PyYAML in /home/stuart/.local/lib/python3.9/site-packages (from blobconverter>=1.2.8->depthai-sdk==1.9.4->-r requirements.txt (line 5)) (6.0.1)
Requirement already satisfied: boto3 in /home/stuart/.local/lib/python3.9/site-packages (from blobconverter>=1.2.8->depthai-sdk==1.9.4->-r requirements.txt (line 5)) (1.28.10)
Requirement already satisfied: botocore<1.32.0,>=1.31.10 in /home/stuart/.local/lib/python3.9/site-packages (from boto3->blobconverter>=1.2.8->depthai-sdk==1.9.4->-r requirements.txt (line 5)) (1.31.10)
Requirement already satisfied: jmespath<2.0.0,>=0.7.1 in /home/stuart/.local/lib/python3.9/site-packages (from boto3->blobconverter>=1.2.8->depthai-sdk==1.9.4->-r requirements.txt (line 5)) (1.0.1)
Requirement already satisfied: s3transfer<0.7.0,>=0.6.0 in /home/stuart/.local/lib/python3.9/site-packages (from boto3->blobconverter>=1.2.8->depthai-sdk==1.9.4->-r requirements.txt (line 5)) (0.6.1)
Requirement already satisfied: python-dateutil<3.0.0,>=2.1 in /home/stuart/.local/lib/python3.9/site-packages (from botocore<1.32.0,>=1.31.10->boto3->blobconverter>=1.2.8->depthai-sdk==1.9.4->-r requirements.txt (line 5)) (2.8.2)
Requirement already satisfied: six>=1.5 in /usr/lib/python3/dist-packages (from python-dateutil<3.0.0,>=2.1->botocore<1.32.0,>=1.31.10->boto3->blobconverter>=1.2.8->depthai-sdk==1.9.4->-r requirements.txt (line 5)) (1.16.0)
Installing collected packages: opencv-contrib-python, depthai
Successfully installed depthai-2.20.2.0 opencv-contrib-python-4.5.5.62
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Ignoring open3d: markers 'platform_machine != "armv6l" and platform_machine != "armv7l" and python_version < "3.9" and platform_machine != "aarch64"' don't match your environment
Requirement already satisfied: ffmpy3==0.2.4 in /home/stuart/.local/lib/python3.9/site-packages (from -r requirements-optional.txt (line 2)) (0.2.4)
Requirement already satisfied: pyusb==1.2.1 in /home/stuart/.local/lib/python3.9/site-packages (from -r requirements-optional.txt (line 3)) (1.2.1)
Requirement already satisfied: sentry-sdk==1.5.1 in /home/stuart/.local/lib/python3.9/site-packages (from -r requirements-optional.txt (line 4)) (1.5.1)
Requirement already satisfied: mcap>=0.0.10 in /home/stuart/.local/lib/python3.9/site-packages (from -r requirements-optional.txt (line 7)) (1.1.0)
Requirement already satisfied: mcap-ros1-support==0.0.8 in /home/stuart/.local/lib/python3.9/site-packages (from -r requirements-optional.txt (line 8)) (0.0.8)
Requirement already satisfied: rosbags==0.9.11 in /home/stuart/.local/lib/python3.9/site-packages (from -r requirements-optional.txt (line 9)) (0.9.11)
Requirement already satisfied: certifi in /usr/lib/python3/dist-packages (from sentry-sdk==1.5.1->-r requirements-optional.txt (line 4)) (2020.6.20)
Requirement already satisfied: urllib3>=1.10.0 in /usr/lib/python3/dist-packages (from sentry-sdk==1.5.1->-r requirements-optional.txt (line 4)) (1.26.5)
Requirement already satisfied: lz4 in /home/stuart/.local/lib/python3.9/site-packages (from rosbags==0.9.11->-r requirements-optional.txt (line 9)) (4.3.2)
Requirement already satisfied: numpy in /home/stuart/.local/lib/python3.9/site-packages (from rosbags==0.9.11->-r requirements-optional.txt (line 9)) (1.25.1)
Requirement already satisfied: ruamel.yaml in /home/stuart/.local/lib/python3.9/site-packages (from rosbags==0.9.11->-r requirements-optional.txt (line 9)) (0.17.32)
Requirement already satisfied: zstandard in /home/stuart/.local/lib/python3.9/site-packages (from rosbags==0.9.11->-r requirements-optional.txt (line 9)) (0.21.0)
Requirement already satisfied: ruamel.yaml.clib>=0.2.7 in /home/stuart/.local/lib/python3.9/site-packages (from ruamel.yaml->rosbags==0.9.11->-r requirements-optional.txt (line 9)) (0.2.7)
stuart@mypi:~/Luxonis/depthai $ python3 depthai_demo.py
Third party libraries failed to import: libhdf5_serial.so.103: cannot open shared object file: No such file or directory
Run "python3 install_requirements.py" to install dependencies
```

stuartcrobinson, 2023-07-25:
i eventually got rid of the `libhdf5_serial.so` warning after trying a few google recommendations, and then i got this error:

`Third party libraries failed to import: libcblas.so.3: cannot open shared object file: No such file or directory`

so then i tried `sudo apt-get install libcblas-dev`, but the terminal advised i run this instead: `sudo apt-get install libatlas-base-dev`

and then the `depthai_demo.py` actually loaded and opened image windows, but it's just a still image, not responsive to the camera after loading.  and the `color` window is blank

[image]

also those input sliders dont do anything.  anything i'm doing wrong?

themarpe, 2023-07-25:
@stuartcrobinson did you install using the:
```
sudo wget -qO- https://docs.luxonis.com/install_depthai.sh | bash
```

command as specified there? And do you mind sharing output of that as well?

Also are you on RPi Buster or Bullseye?

stuartcrobinson, 2023-07-26:
@themarpe yes i believe so, but i've since wiped that OS after discovering that u provide your own depthai raspberry pi OS that i'm trying now.  

>Also are you on RPi Buster or Bullseye?

not sure, it's a pi 4 and i just ran whatever the default was here https://www.raspberrypi.com/software/

stuartcrobinson, 2023-07-26:
closing as the original issue was resolved and the blank color camera feed seems to be a hardware issue 
https://github.com/luxonis/depthai/issues/1090#issuecomment-1651918881
https://github.com/luxonis/depthai/issues/1085#issuecomment-1651921020