luxonis/depthai #1090: [BUG] rasp pi depthai image OAK_CM4_POE_v9.7: `color` camera not displaying in `depthai_demo.py`

stuartcrobinson, 2023-07-26:
running OAK_CM4_POE_v9.7 on a raspberry pi 4

`python depthai_demo.py` opens the full demo application thing but the camera window says `Starting demo...` forever while the `color` option is selected.  

other options like `depth`, `left`, etc seem to work.  but they're all still images, maybe that's expected.  but i can never get anythnig from the `color` camera

[image]

[image]

could this be a power issue also?  like u mentioned here https://github.com/luxonis/depthai/issues/1085#issuecomment-1650662308

Erol444, 2023-07-26:
Hi @stuartcrobinson ,
For the 1st issue (`'depthai.CameraBoardSocket' has no attribute 'CAM_A'`), you are using an old version of depthai library. Update with `python3 -mpip install depthai -U`).
On the other issue you printed cam_test already, and it indeed is a HW issue. Please contact supprot@luxonis.com with the link to that issue, and we will replace the device asap.

closed 2023-07-26 by Erol444

stuartcrobinson, 2023-07-26:
for completeness/comparison, here's the output after updating depthai

and unlike on my mac, this actually brought up 4 camera windows.  `left` and `right` showed live video feeds, while `rgb` and `camd` were blank black frames

```
pi@luxonis:~/repos/depthai-python/utilities $ python3 cam_test.py -rs
Enabled cameras:
    rgb : color
   left : mono
  right : mono
   camd : color
DepthAI version: 2.22.0.0
DepthAI path: /home/pi/.local/lib/python3.9/site-packages/depthai.cpython-39-arm-linux-gnueabihf.so
/home/pi/repos/depthai-python/utilities/cam_test.py:284: DeprecationWarning: Use constructor taking 'UsbSpeed' instead
  with dai.Device(*dai_device_args) as device:
[184430109175610E00] [1.1.1] [0.940] [ColorCamera(9)] [error] Camera not detected on socket: 3
[184430109175610E00] [1.1.1] [0.941] [MonoCamera(7)] [error] OV7251 only supports THE_480_P/THE_400_P resolutions, defaulting to THE_480_P
[184430109175610E00] [1.1.1] [0.941] [MonoCamera(5)] [error] OV7251 only supports THE_480_P/THE_400_P resolutions, defaulting to THE_480_P
Connected cameras:
 -socket CAM_A : IMX214 4208 x 3120 focus:auto  - COLOR
 -socket CAM_B : OV7251  640 x  480 focus:fixed - MONO
 -socket CAM_C : OV7251  640 x  480 focus:fixed - MONO
USB speed: SUPER
IR drivers: []
Cam:      rgb           left          right         camd     [host | capture timestamp]
FPS:   0.00|  0.00  30.06| 30.01  29.82| 30.01   0.00|  0.00
```