luxonis/depthai #1091: [BUG] {OAK-D Calibration Failure}

aiManny, 2023-07-26:
Yesterday I tried running the calibration script for my OAK-D with `python3 calibrate.py -s 2.2 -brd bw1098obc -db`. I successfully captured all images and viewed reprojection error however but was then interrupted by this error:

 [image]
My setup uses:

- **VMWare Ubuntu 20.04 virtual machine**
- **OAK-D camera (from original kickstart campaign)**
- **depthai 2.22.0.0**

@jakaskerl alluded to possible issues with the script itself. Any thoughts @MaticTonin?
I'm recalibrating to try fixing my depth pipeline failure described on this [thread](https://discuss.luxonis.com/d/1702-fatal-error-on-mss-cpu-trap-09-address-802ce934-0/14)

Erol444, 2023-07-26:
Hi @aiManny ,
Could you run https://docs.luxonis.com/projects/api/en/latest/samples/calibration/calibration_reader/#calibration-reader and paste the generated .json here?

aiManny, 2023-07-27:
Do I need a specific version of the depthai library? I tried to run the calibration reader but got: 

[image]

I edited the script to read the JSON from my successful calibration I did a year ago

themarpe, 2023-07-28:
@aiManny 

Use latest depthai library (v2.22.0) and then run examples/calibration/calibration_dump.py example and paste the output

aiManny, 2023-07-28:
Here's the raw output:

Is EEPROM available: True
No user calibration: Could not allocate string object!
No factory calibration: No or invalid EEPROM configuration flashed, error: EEPROM_INVALID_DATA
User calibration raw: [5, 0, 170, 85, 3, 0, 0, 0, 0, 66, 87, 49, 48, 57, 56, 79, 66, 67, 0, 0, 0, 0, 0, 0, 0, 82, 48, 77, 48, 69, 48, 0, 0, 0, 0, 83, 255, 127, 63, 53, 106, 23, 59, 226, 9, 128, 59, 201, 188, 23, 187, 198, 255, 127, 63, 150, 146, 164, 58, 223, 226, 127, 187, 183, 193, 165, 186, 115, 255, 127, 63, 197, 251, 127, 63, 218, 63, 32, 60, 55, 120, 189, 187, 132, 94, 32, 188, 207, 252, 127, 63, 41, 51, 162, 186, 82, 16, 189, 59, 140, 155, 169, 58, 219, 254, 127, 63, 1, 2, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 255, 32, 3, 0, 5, 0, 25, 195, 87, 68, 0, 0, 0, 0, 188, 116, 30, 68, 0, 0, 0, 0, 22, 117, 87, 68, 126, 77, 195, 67, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 128, 63, 178, 90, 193, 65, 74, 58, 216, 66, 172, 247, 46, 187, 114, 199, 115, 186, 40, 249, 179, 67, 167, 217, 195, 65, 248, 218, 206, 66, 82, 172, 185, 67, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 82, 184, 143, 66, 0, 1, 26, 251, 127, 63, 26, 70, 245, 187, 75, 88, 30, 60, 6, 119, 244, 59, 245, 253, 127, 63, 12, 151, 41, 59, 70, 168, 30, 188, 32, 218, 36, 187, 184, 252, 127, 63, 239, 192, 241, 192, 98, 87, 151, 189, 236, 239, 50, 61, 0, 0, 240, 192, 0, 0, 0, 0, 0, 0, 0, 0, 2, 32, 3, 0, 5, 0, 222, 180, 86, 68, 0, 0, 0, 0, 244, 190, 30, 68, 0, 0, 0, 0, 17, 148, 86, 68, 208, 40, 194, 67, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 128, 63, 193, 215, 49, 193, 29, 192, 78, 66, 51, 97, 41, 187, 20, 101, 189, 186, 5, 119, 101, 194, 122, 227, 49, 193, 66, 95, 78, 66, 61, 114, 100, 194, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 82, 184, 143, 66, 0, 2, 232, 251, 127, 63, 51, 9, 48, 60, 56, 41, 73, 187, 99, 134, 48, 188, 0, 249, 127, 63, 84, 218, 33, 188, 249, 46, 66, 59, 116, 98, 34, 60, 127, 252, 127, 63, 79, 33, 112, 64, 62, 102, 38, 187, 139, 38, 19, 189, 0, 0, 112, 64, 0, 0, 0, 0, 0, 0, 0, 0, 0, 56, 4, 128, 7, 135, 148, 113, 185, 68, 0, 0, 0, 0, 228, 182, 110, 68, 0, 0, 0, 0, 131, 53, 185, 68, 194, 161, 0, 68, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 128, 63, 106, 150, 254, 64, 173, 20, 53, 195, 72, 219, 196, 187, 216, 75, 15, 187, 251, 147, 157, 68, 46, 74, 241, 64, 103, 38, 49, 195, 44, 108, 155, 68, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 109, 150, 137, 66, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 25 [... cut ...]  255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255]

themarpe, 2023-07-28:
@aiManny add `--noInitCalibration` argument and you might also have to apply a protected flashing value to override. `DEPTHAI_ALLOW_FACTORY_FLASHING=235539980`. Let me know if this works for you.

Note, this combination will remove all existing calibration/eeprom info, so use with care

aiManny, 2023-07-28:
Is the idea that removing the existing calibration will allow for a smooth recalibration? I ask because as of right now I don't have a way to recalibrate my device if I were to loose it's current eeprom info.

themarpe, 2023-07-29:
@aiManny correct + the original calibration is still here in the post so coukd be reconstructed if need may be

aiManny, 2023-07-31:
I get the same console output even with the `--noInitCalibration` and `DEPTHAI_ALLOW_FACTORY_FLASHING=235539980` arguments

aiManny, 2023-08-03:
I also downgraded to `depthai 2.5.0.0` and retried the old calibration script I'd used last summer, however I got the same error: `UnicodeDecodeError: 'utf-8' codec can't decode byte 0xa4 in position 0: invalid start byte`

aiManny, 2023-08-03:
The calibration flash was successful. When I try running a depth preview I'm no longer seeing a CPU trap error, but I'm getting this instead: 

`[14442C10619FF0D000] [3.2.1] [1.373] [StereoDepth(4)] [error] There is no available extrinsic calibration between camera ID: '1' and '2'`

`Traceback (most recent call last):
  File "depth_test.py", line 149, in <module>
    stereo_preview(pipeline, disparity)
  File "depth_test.py", line 89, in stereo_preview
    inDisparity = q.get()  # blocking call, will wait until a new data has arrived
RuntimeError: Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'Depth' (X_LINK_ERROR)'`

Did that `json` have calibration data for the stereo pair as well?

themarpe, 2023-08-04:
@aiManny 
No, it didn't include calibration data. It just resolved the weird state of the eeprom. Now continue with calibrate.py as in the beginning and it should work as expected now

closed 2023-08-04 by themarpe