luxonis/depthai #1095: Demo script not working on OAK-D

kXborg, 2023-08-06:
Using windows 11.
Python=3.11
depthai 2.21.2.0
depthai-sdk 1.9.4

The following error is observed. 

```
Using depthai module from:  C:\Users\Kukil\AppData\Roaming\Python\Python311\site-packages\depthai.cp311-win_amd64.pyd
Depthai version installed:  2.21.2.0
Setting up demo...
Available devices:
[0] 14442C10B1E393D000 [X_LINK_UNBOOTED]
USB Connection speed: UsbSpeed.SUPER
[14442C10B1E393D000] [2.2.3] [1.161] [StereoDepth(9)] [error] RGB camera calibration missing, aligning to RGB won't work
F: [global] [         0] [ThreadN] dispatcherResponseServe:925  no request for this response: XLINK_WRITE_RESP 1

Stopping demo...
=== TOTAL FPS ===
[color]: 29.3
[left]: 0.0
[right]: 0.0
  File "C:\Users\Kukil\Desktop\depthai\depthai_demo.py", line 656, in run
    self.instance.run_all(self.conf)
  File "C:\Users\Kukil\Desktop\depthai\depthai_demo.py", line 133, in run_all
    self.run()
  File "C:\Users\Kukil\Desktop\depthai\depthai_demo.py", line 317, in run
    self.loop()
  File "C:\Users\Kukil\Desktop\depthai\depthai_demo.py", line 373, in loop
    self._pv.prepareFrames(callback=self.onNewFrame)
  File "C:\Users\Kukil\AppData\Roaming\Python\Python311\site-packages\depthai_sdk\managers\preview_manager.py", line 148, in prepareFrames
    packet = queue.tryGet()
             ^^^^^^^^^^^^^^
RuntimeError: Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'color' (X_LINK_ERROR)'
```
Please help.

Erol444, 2023-08-08:
Hi @kXborg , it seems like you have quite an old device, which doesn't have RGB calibration. Could you try [recalibrating the device](https://docs.luxonis.com/projects/hardware/en/latest/pages/guides/calibration/)?

kXborg, 2023-08-09:
Hi @Erol444,

Thanks, it is working after calibration. There were a few hiccups, I am noting them down here in case someone faces the issue.
1. Submodules were not getting cloned (had to clone them individually) `depthai_calibration` and `resources/depthai_boards`.
2. Using the laptop screen for displaying ChAruco board is not a good idea. Good luck orienting that. 
3. Minimum size of one square should be 2.2 cm. If your iPad / tablet is big enough, it might work. Mine is just 10.4 inch (squares were 1.4 cm). Got away with a chopping board and an A4 sheet finally.
4. It took me 29 captures for complete calibration.