luxonis/depthai #1105: [BUG] {Image artifacts when using onboard encoding on FFC-6P and RCV3}

JordanGell, 2023-08-23:
Dear Luxonis, we've been having the following issue with the FFC-6P 

**Hardware**

-OAK FFC-6P
-socket CAM_B : OV9282  640 x  480 focus:fixed - MONO
-socket CAM_C : OV9282  640 x  480 focus:fixed - MONO
-socket CAM_F : IMX412 4056 x 3040 focus:fixed - COLOR

**Software** 
-DepthAI version: 2.22.0.0
-Branch RCV3_Develop

**Description**

When using the onboard video encoding on the FFC-6P we get good frames with a stationary subject however with any motion of the subject artefacts appear (Example video attached). We are also unable to get video from the IR cameras when using the onboard video encoding, only empty files.

**Expected behavior**

Frames encoded on the FFC-6P for all three cameras without artefacts 

**Attach system log**
[log_system_information.txt](https://github.com/luxonis/depthai/files/12421016/log_system_information.txt)

**Video**
https://github.com/luxonis/depthai/assets/104423166/ecf4a90b-4a2e-472c-8eb0-c8da503931d8

brmarkus, 2023-08-23:
Can you maybe take one of the encoded (non-empty, non-black) frames and highlight artefacts, please?
It might be my video-player, or I just missed the important frames - I only see "noise" due to the encoder using a low bitrate (and/or operating in low-latency mode).

Example:
[image]

JordanGell, 2023-08-24:
We use the auto bitrate setting, with the latest 6P firmware, and we do not turn on low-latency (what is this setting?) there are a couple of bits that look like artifcacts but mainly it's the low quality output or noise that we're concerned with. We do not get this behaviour when not using the Video Encoder. This is using the H264 presets - when using the HEVC, the quality is worse and when using MJPEG we recieve one (clear good quality) frame then the entire pipeline locks up.