luxonis/depthai #1107: How can i chagne the based mono-camera for auto-exposure?

socome, 2023-08-29:
Hi. 

When I use the Oak-D Pro, it appears that the auto-exposure of the mono cameras is functioning based on the right mono camera. How can I switch the base mono-camera from right to left? Additionally, is it possible to set auto-exposure for the mono cameras separately?

socome, 2023-08-30:
@Erol444

Erol444, 2023-08-30:
Hi @socome ,
I am not completely sure, but I don't think it's possible to change that via API. For AE, you can also change exposure time/sensitivity manually for each camera separately, examples:
- [SDK example](https://docs.luxonis.com/projects/sdk/en/latest/samples/CameraComponent/sdk_camera_control/#camera-control)
- [depthai API example](https://docs.luxonis.com/projects/api/en/latest/samples/ColorCamera/rgb_camera_control/#rgb-camera-control)