luxonis/depthai #1128: [BUG] Dependency installation issues

camchaney, 2023-10-16:
I am trying to install the dependencies on my Raspberry Pi 4 running 32 bit OS using `sudo curl -fL https://docs.luxonis.com/install_dependencies.sh | bash`. The error log I get is pasted below. I initially tried this with the 64 bit Raspbian OS but got the same error and thought it might be due to incompatibility with the new architecture. So I reflashed 32 bit and tried again, but it didn't work. I also added `arm_64bit=0` to the config.txt file to make sure that only the 32 bit OS is being used. It's weird because I got this installation to run with no problems on my old Raspberry Pi 32 bit image, on a different older SD card.

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:05 --:--:--     0
100  5671  100  5671    0     0    979      0  0:00:05  0:00:05 --:--:--   979
Hit:1 http://archive.raspberrypi.com/debian bookworm InRelease
Hit:2 http://raspbian.raspberrypi.com/raspbian bookworm InRelease
Reading package lists... Done
W: http://raspbian.raspberrypi.com/raspbian/dists/bookworm/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package libtbb2 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  libtbbmalloc2

E: Package 'libtbb2' has no installation candidate
E: Unable to locate package libdc1394-22-dev

Failed installing dependencies. Could be a bug in the installer or unsupported platform. Open a bug report over at https://github.com/luxonis/depthai - exited with status 100 at line 123

brmarkus, 2023-10-16:
Others seems to have the same finding, like under "https://discuss.luxonis.com/d/1999-debian-bookworm-oak-1".

camchaney, 2023-10-16:
Awesome, thanks. I followed this but got another error:

error: externally-managed-environment

This environment is externally managed

To install Python packages system-wide, try apt install
python3-xyz, where xyz is the package you are trying to
install.

If you wish to install a non-Debian-packaged Python package,
create a virtual environment using python3 -m venv path/to/venv.
Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
sure you have python3-full installed.

For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

Failed installing dependencies. Could be a bug in the installer or unsupported platform. Open a bug report over at https://github.com/luxonis/depthai - exited with status 1 at line 120

brmarkus, 2023-11-27:
Did you try one of the notes mentioned above in this thread?

LaudixGit, 2023-11-27:
Nope. This script is failing as-is and I reported it as such.
(so the next person looking will know the script has not been repaired yet)

jnpm, 2023-12-29:
> Awesome, thanks. I followed this but got another error:
>
> error: externally-managed-environment
>
> This environment is externally managed
>
> To install Python packages system-wide, try apt install [...]

I have the same issue. Have you resolved it?

katerina-stemberova, 2024-02-22:
> error: externally-managed-environment

I believe I have a solution for this issue. The problem arises when `pip` is told to upgrade itself at the end of the installation of the dependencies. If `pip` was installed as an OS package (which seems to be [the case](https://packages.debian.org/stable/python/python3-pip) at least with default Python installations on Debian), it will throw the error because it's not supposed to install/manage packages system-wide. Since we just want it to upgrade itself (and not install anything), I just added an alternative command for doing so using `apt-get` in environments that are externally managed (in the "non-external" ones it will run the original command).

@camchaney @jnpm would you mind testing the [updated script](https://github.com/katerina-stemberova/depthai-python/blob/1128_install-dependencies/docs/source/_static/install_dependencies.sh)? Just download the file and run it locally (`bash install_dependencies.sh`). It works nicely for me, but it would be good to get a confirmation to avoid the "works on my machine" effect :)

For reference: this behavior was implemented in Python 3.10.

katerina-stemberova, 2024-02-23:
The [newest version of the script](https://github.com/katerina-stemberova/depthai-python/blob/1128_install-dependencies/docs/source/_static/install_dependencies.sh) should also fix the issue with installation of the "libtbb2" package.

@camchaney @LaudixGit feel free to test (download the file and run it locally - `bash install_dependencies.sh`)

LaudixGit, 2024-05-06:
> The [newest version of the script](https://github.com/katerina-stemberova/depthai-python/blob/1128_install-dependencies/docs/source/_static/install_dependencies.sh) should also fix the issue with installation of the "libtbb2" package.

Writing to confirm this version did execute correctly and I am able to successfully run examples on RPi5