luxonis/depthai #1129: [BUG] {python3 install_requirements.txt failing}

dmorris0, 2023-10-18:
I have a RPi 4 with Ubuntu Jammy 22.04 and an Oak-D Lite connected through the USB 3 connector.  I cloned the depthai github repo and am stepping through the installation.  Everything seems to work until I get to the install_requirements.py line:

```
ubuntu@tbot:~/depthai$ python3 install_requirements.py
pip 23.3 from /home/ubuntu/.local/lib/python3.10/site-packages/pip (python 3.10)
Requirement already satisfied: pip in /home/ubuntu/.local/lib/python3.10/site-packages (23.3)
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff9f2197b0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/pip/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff9f219960>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/pip/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff9f219e10>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/pip/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff9f219fc0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/pip/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffff9f21a170>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/pip/
WARNING: Skipping opencv-python as it is not installed.
WARNING: Skipping opencv-contrib-python as it is not installed.
WARNING: Skipping depthai as it is not installed.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://artifacts.luxonis.com/artifactory/luxonis-depthai-data-local/wheels/, https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local/
Ignoring pyqt5: markers 'platform_machine != "armv6l" and platform_machine != "armv7l" and platform_machine != "aarch64" and platform_machine != "arm64"' don't match your environment
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffa667abc0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/requests/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffa667aec0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/requests/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffa667b070>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/requests/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffa667b220>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/requests/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffa667b3d0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/requests/
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffa667b850>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/requests/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffa667bac0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/requests/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffa6679f60>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/requests/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffa667a950>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/requests/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffa667afb0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/requests/
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffa667abc0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /artifactory/luxonis-depthai-data-local/wheels/requests/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffa667be20>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /artifactory/luxonis-depthai-data-local/wheels/requests/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffa667bfd0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /artifactory/luxonis-depthai-data-local/wheels/requests/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffa66a81c0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /artifactory/luxonis-depthai-data-local/wheels/requests/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffa66a8370>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /artifactory/luxonis-depthai-data-local/wheels/requests/
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffa66a85b0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /artifactory/luxonis-python-snapshot-local/requests/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0xffffa66a8880>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /artifactory/luxonis-python-snapshot-local/requests/
[... cut ...]
  File "/home/ubuntu/depthai/install_requirements.py", line 44, in <module>
    subprocess.check_call(pip_package_install + ["-r", "requirements.txt"], cwd=script_directory)
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/python3', '-m', 'pip', 'install', '-U', '--prefer-binary', '--user', '-r', 'requirements.txt']' returned non-zero exit status 1.
```

I'm not sure what to expect, but this isn't correct.  Any advice one what to do?

brmarkus, 2023-10-18:
When you see this failing - do you still have an Internet connection, can access other online resources?
Seems a network connection could not be established anylonger.

Could it be a power-supply issue (while installing the SoC is heating up, a lot of processing power required), or simply a loose connection?

dmorris0, 2023-10-18:
There seems to be some connection problem, although now with the USB.  When I run the first demo, it initially works (I see the depth and color images), and then after a couple seconds it fails:
```
python3 depthai_demo.py -gt cv
Using depthai module from:  /home/ubuntu/.local/lib/python3.10/site-packages/depthai.cpython-310-aarch64-linux-gnu.so
Depthai version installed:  2.21.2.0
Setting up demo...
Available devices:
[0] 19443010817AF71200 [X_LINK_UNBOOTED]
USB Connection speed: UsbSpeed.SUPER
Stopping demo...
=== TOTAL FPS ===
[color]: 6.4
[disparity]: 6.4
[disparityColor]: 6.4
[nn]: 0.0
Traceback (most recent call last):
  File "/home/ubuntu/depthai/depthai_demo.py", line 1053, in <module>
    runOpenCv()
  File "/home/ubuntu/depthai/depthai_demo.py", line 1043, in runOpenCv
    demo.run_all(confManager)
  File "/home/ubuntu/depthai/depthai_demo.py", line 133, in run_all
    self.run()
  File "/home/ubuntu/depthai/depthai_demo.py", line 317, in run
    self.loop()
  File "/home/ubuntu/depthai/depthai_demo.py", line 373, in loop
    self._pv.prepareFrames(callback=self.onNewFrame)
  File "/home/ubuntu/.local/lib/python3.10/site-packages/depthai_sdk/managers/preview_manager.py", line 148, in prepareFrames
    packet = queue.tryGet()
RuntimeError: Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'color' (X_LINK_ERROR)'
```

dmorris0, 2023-10-18:
It is using a battery, but no harsh conditions.  I might be able to try a different power supply.  I could also try some heavy jobs, although not sure how to measure stability.

Jai-GAY, 2024-08-28:
python3 ~/Luxonis/depthai/install_requirements.py
subprocess.CalledProcessError: Command '['/usr/bin/python3', '-m', 'pip', 'install', '-U', '--prefer-binary', '--user', '-r', 'requirements.txt']' returned non-zero exit status 2.
```
$ uname -a
Linux pi-ubuntu-desktop 5.15.0-1048-raspi #51-Ubuntu SMP PREEMPT Thu Feb 22 10:30:12 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux
$ sudo lsb_release -a
[sudo] password for pi: 
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.4 LTS
Release:	22.04
Codename:	jammy
```
```
python3 ~/Luxonis/depthai/depthai_demo.py
Third party libraries failed to import: No module named 'xmltodict'
Run "python3 install_requirements.py" to install dependencies
python3 ~/Luxonis/depthai/install_requirements.py
pip 24.2 from /home/pi/.local/lib/python3.10/site-packages/pip (python 3.10)
Requirement already satisfied: pip in ./.local/lib/python3.10/site-packages (24.2)
Found existing installation: opencv-python 4.10.0.84
Uninstalling opencv-python-4.10.0.84:
  Successfully uninstalled opencv-python-4.10.0.84
WARNING: Skipping opencv-contrib-python as it is not installed.
Found existing installation: depthai 2.28.0.0
Uninstalling depthai-2.28.0.0:
  Successfully uninstalled depthai-2.28.0.0
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://artifacts.luxonis.com/artifactory/luxonis-depthai-data-local/wheels/, https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local/
Ignoring pyqt5: markers 'platform_machine != "armv6l" and platform_machine != "armv7l" and platform_machine != "aarch64" and platform_machine != "arm64"' don't match your environment
Collecting requests==2.26.0 (from -r requirements.txt (line 1))
  Downloading https://www.piwheels.org/simple/requests/requests-2.26.0-py2.py3-none-any.whl (62 kB)
Requirement already satisfied: numpy>=1.21.4 in /home/pi/.local/lib/python3.10/site-packages (from -r requirements.txt (line 3)) (2.1.0)
Collecting opencv-contrib-python==4.5.5.62 (from -r requirements.txt (line 4))
  Downloading opencv_contrib_python-4.5.5.62-cp36-abi3-manylinux_2_17_aarch64.manylinux2014_aarch64.whl.metadata (18 kB)
Collecting depthai-sdk==1.9.4 (from -r requirements.txt (line 5))
  Downloading https://www.piwheels.org/simple/depthai-sdk/depthai_sdk-1.9.4-py3-none-any.whl (167 kB)
Collecting depthai==2.24.0.0 (from -r requirements.txt (line 9))
  Downloading depthai-2.24.0.0-cp310-cp310-manylinux_2_17_aarch64.manylinux2014_aarch64.whl.metadata (8.7 kB)
Collecting Qt.py (from -r requirements.txt (line 10))
  Downloading https://www.piwheels.org/simple/qt-py/Qt.py-1.4.1-py2.py3-none-any.whl (35 kB)
Requirement already satisfied: scipy in /home/pi/.local/lib/python3.10/site-packages (from -r requirements.txt (line 11)) (1.13.1)
Collecting scipy (from -r requirements.txt (line 11))
  Downloading scipy-1.14.1-cp310-cp310-manylinux_2_17_aarch64.manylinux2014_aarch64.whl.metadata (60 kB)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in /usr/lib/python3/dist-packages (from requests==2.26.0->-r requirements.txt (line 1)) (1.26.5)
Requirement already satisfied: certifi>=2017.4.17 in /usr/lib/python3/dist-packages (from requests==2.26.0->-r requirements.txt (line 1)) (2020.6.20)
Collecting charset-normalizer~=2.0.0 (from requests==2.26.0->-r requirements.txt (line 1))
  Downloading https://www.piwheels.org/simple/charset-normalizer/charset_normalizer-2.0.12-py3-none-any.whl (44 kB)
Requirement already satisfied: idna<4,>=2.5 in /usr/lib/python3/dist-packages (from requests==2.26.0->-r requirements.txt (line 1)) (3.3)
Collecting blobconverter>=1.2.8 (from depthai-sdk==1.9.4->-r requirements.txt (line 5))
  Downloading https://www.piwheels.org/simple/blobconverter/blobconverter-1.4.3-py3-none-any.whl (10 kB)
Collecting pytube>=12.1.0 (from depthai-sdk==1.9.4->-r requirements.txt (line 5))
  Downloading https://www.piwheels.org/simple/pytube/pytube-15.0.0-py3-none-any.whl (57 kB)
Collecting PyTurboJPEG==1.6.4 (from depthai-sdk==1.9.4->-r requirements.txt (line 5))
  Downloading https://www.piwheels.org/simple/pyturbojpeg/PyTurboJPEG-1.6.4-py3-none-any.whl (13 kB)
Collecting marshmallow==3.17.0 (from depthai-sdk==1.9.4->-r requirements.txt (line 5))
  Downloading https://www.piwheels.org/simple/marshmallow/marshmallow-3.17.0-py3-none-any.whl (48 kB)
Collecting distinctipy (from depthai-sdk==1.9.4->-r requirements.txt (line 5))
  Downloading https://www.piwheels.org/simple/distinctipy/distinctipy-1.3.4-py3-none-any.whl (26 kB)
Collecting xmltodict (from depthai-sdk==1.9.4->-r requirements.txt (line 5))
  Downloading https://www.piwheels.org/simple/xmltodict/xmltodict-0.13.0-py2.py3-none-any.whl (12 kB)
Requirement already satisfied: packaging>=17.0 in /usr/lib/python3/dist-packages (from marshmallow==3.17.0->depthai-sdk==1.9.4->-r requirements.txt (line 5)) (21.3)
Collecting types-PySide2 (from Qt.py->-r requirements.txt (line 10))
  Downloading https://www.piwheels.org/simple/types-pyside2/types_pyside2-5.15.2.1.7-py2.py3-none-any.whl (572 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 572.2/572.2 kB 538.4 kB/s eta 0:00:00
Requirement already satisfied: PyYAML in /home/pi/.local/lib/python3.10/site-packages (from blobconverter>=1.2.8->depthai-sdk==1.9.4->-r requirements.txt (line 5)) (6.0.2)
Downloading opencv_contrib_python-4.5.5.62-cp36-abi3-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (44.8 MB)
   ━╸                                       2.1/44.8 MB 3.1 MB/s eta 0:00:14
[... cut ...]
Traceback (most recent call last):
  File "/home/pi/Luxonis/depthai/install_requirements.py", line 44, in <module>
    subprocess.check_call(pip_package_install + ["-r", "requirements.txt"], cwd=script_directory)
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/python3', '-m', 'pip', 'install', '-U', '--prefer-binary', '--user', '-r', 'requirements.txt']' returned non-zero exit status 2.
```