luxonis/depthai #1137: exited with status 100 at line 216[BUG] {Title of the bug}

abheekmondal, 2023-11-09:
**Check if issue already exists**

- Google it (e.g. *error xy* github luxonis depthai)
- Check [troubleshooting](https://docs.luxonis.com/en/latest/pages/troubleshooting/) in documentation.

**Describe the bug**
A clear and concise description of what the bug is.

**Minimal Reproducible Example**
Append the MRE to the bug report, [instructions here](https://docs.luxonis.com/en/latest/pages/support/#creating-minimal-reproducible-example)

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Pipeline Graph**

Please also provide a screenshot of your pipeline using the [DepthAI Pipeline Graph](https://github.com/geaxgx/depthai_pipeline_graph).

**Attach system log**
 - Provide output of [log_system_information.py](https://github.com/luxonis/depthai/blob/main/log_system_information.py)

**Additional context**
Add any other context about the problem here.

brmarkus, 2023-11-09:
Can you provide more details, please?
Which camera model are you using, connected to which system, which HW, which OperatingSystem?
Which application is causing the problem, how have you started it, using which settings, which model?

abheekmondal, 2023-11-09:
Hi, we are teying to connect ros2bridge to rtmaps so we can work with it on
our Autera. I am attaching an image of the software of our computer.