luxonis/depthai #1139: [BUG] {wrong image rectification using stereo StereoDepth on oak-d-lite}

chameau5050, 2023-11-19:
**Describe the bug**
Hi, I'm trying to calibrate an oak-d-lite in an underwater enclosure. Calibration and flashing the device are working well using your API, but when I try to run the same image rectification on board, the images are not corrected the same way, leading to bad depth perception
 
Bellow the result from calibration API:

[image]

Bellow the rectified image after flashing the calibration on oak-d-lite 

[image]

Any idea why the two rectification are not the same ?

brmarkus, 2023-11-19:
Not sure I understood you right - are you calibrating outside the water (and comparing the results from images under water)?

chameau5050, 2023-11-20:
Sorry for confusion, we are calibrating the camera directly in the enclosure underwater. 

In sumery I do the following step : 

1. I calibrate the camera using the `calibrate.py` script
2. Then the calibration succed and give the following result : 
[image]

3. Then the `calibrate.py` script flash the computed calibration data in the EPROM of my oak-d-lite
4. Once the data is flash I close the `calibrate.py` script
5. Then I start my application that create a simple stereo depth pipeline similar to  [documentation](https://docs.luxonis.com/projects/api/en/latest/components/nodes/stereo_depth/)
6. Then i get the following images from the rectifiedLeft and rectifiedRight output of the StereoDepth node 
[image]

We see that the result of rectification in `calibrate.py`(step 2) are realy curved in the border due to the enclosure effect. But the probleme we have is that the rectification compute on the oak-d-lite is not curved at all (step 6). So my conclusion is that the StereoDepth node is not appling all the image correction it should be doing. 

Any idea why ?

chameau5050, 2023-11-26:
The issue was that I needed to turn on the `enableDistortionCorrection` option of the stereo node like this 

`stereo->enableDistortionCorrection(true);`

closed 2023-11-26 by chameau5050