luxonis/depthai #1142: [BUG] {Depth quality difference between OAK-D-LITE models}

MoscowskyAnton, 2023-11-28:
Hi! 
I am working with OAK-D-Lite cameras and found that depth quality is different between cameras.
I have some cameras that my team won in OpenCV AI competition, and some we bought after via aliexpress retailer.
And here is difference on its depth images (obtained with ROS) on default settings:
[image]
Also I provide mono images of same place
[image]

On the left is new camera (been bought) and on the right is old (been won). Also the ranges in old depth is not seems to be very accurate in comparison to new.
Is there a way to improve quality of olds cameras? Thanks!

brmarkus, 2023-11-28:
Isn't the photo on the right over-exposed? There is a (much) brighter light-source in the door - will be harder for feature-extraction.

MoscowskyAnton, 2023-11-28:
@brmarkus I tried many different places, always the story like this

brmarkus, 2023-11-28:
Where the cameras calibrated before?
Both models are the same, i.e. both are auto-focus or both fixed-focus?
Clean lense?

MoscowskyAnton, 2023-11-28:
@brmarkus 
I don't know where calibration was conducted, I didn't do any (should I? where can I get instruction?)
Also not sure about focus, how can I check this?
Lens looks good, and EVERY of three 'old' cams works in the same manner

themarpe, 2023-12-07:
@MoscowskyAnton You'll have to update the depthai version to latest. `python3 -m pip install -U depthai`. Its likely too old version