luxonis/depthai #1160: [BUG] Unable to deploy Roboflow model to Oak Pro D

avitus, 2024-03-19:
[==================================================]
Done
USB Speed: UsbSpeed.HIGH
None CAM_A_video 0
None 7_disparity 0
[19443010B1359D2E00] [0.1] [138.381] [VideoEncoder(1)] [critical] Out of memory while creating pool for 'bitstream' frames. Number of frames: 4 each with size: 3170304B
[19443010B1359D2E00] [0.1] [138.443] [VideoEncoder(8)] [critical] Out of memory while creating pool for 'bitstream' frames. Number of frames: 4 each with size: 444928B
[19443010B1359D2E00] [0.1] [143.021] [DetectionNetwork(10)] [critical] Fatal error in openvino 'universal'. Likely because the model was compiled for different openvino version. If you want to select an explicit openvino version use: setOpenVINOVersion while creating pipeline. If error persists please report to developers. Log: 'Convolution' '265' 'CMX memory is not enough!'.
[19443010B1359D2E00] [0.1] [146.491] [system] [critical] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 00, address: 00000000' '0'
[2024-03-18 20:31:44] INFO [root.close:456] Closing OAK camera

Erol444, 2024-04-01:
Hi @avitus ,
Could you share the **minimal** repro example together with the model, so we can try to repro?

avitus, 2024-04-16:
Sorry for the slow reply. Here's the code

```
from depthai_sdk import OakCamera, RecordType
from depthai_sdk.trigger_action.actions.record_action import RecordAction
from depthai_sdk.trigger_action.triggers.detection_trigger import DetectionTrigger
from roboflow import Roboflow

rf = Roboflow(api_key="xxx")

workspaceId = 'avitus'
projectId = 'family-id'
rf_project = rf.workspace(workspaceId).project(projectId)

with OakCamera() as oak:

    color = oak.create_camera('color', resolution='128p', fps=3, encode='H264')

    model_config = {
        'source': 'roboflow', 
        'model':'family-id/4',  
        'key':'xxx' # Roboflow API key
    }

    nn = oak.create_nn(model_config, color)

    trigger = DetectionTrigger(input=nn, min_detections={'Portobello': 1}, cooldown=60)

    action = RecordAction(inputs=[color], 
                          dir_path='./vitus_recordings/',
                          duration_before_trigger=1, 
                          duration_after_trigger=6)

    oak.trigger_action(trigger=trigger, action=action)
    oak.visualize(color)
    oak.start(blocking=True)
```

avitus, 2024-04-16:
Is there a way I can get you the API key?

Erol444, 2024-04-17:
Hi @avitus ,
You could send the API key to erik@luxonis.com and I can take a look at it. But from error message it's likely because the model is too large (`Log: 'Convolution' '265' 'CMX memory is not enough!'.`), in which case I can't do much.