luxonis/depthai #1162: [BUG] NameError: name 'EncodingQuality' is not defined

flipphillips, 2024-03-23:
**Check if issue already exists**

There are references in other repos - https://github.com/luxonis/depthai-experiments/issues/478

**Describe the bug**

Attempting to run `./depthai_demo.py -app record` fails after building its `venv` with `NameError: name 'EncodingQuality' is not defined`

This happens whether using the `venv` that gets built in `apps/record` (which builds with a different error with version of `av`) or using the 'base' `venv` built using the `requirements.txt` in the root. 

**Minimal Reproducible Example**

`./depthai_demo.py -app record`

also

❯ cd apps/record

❯ mv requirements.txt requirements.SAV (* to use the environment from the root, or leave it, it won't build because of bugs in `av`, you can update the version of `av` so it will build, but then the same error appears. *)

❯ python ./main.py
Traceback (most recent call last):
  File "/home/flip/GitHub/depthai/apps/record/./main.py", line 13, in <module>
    **from depthai_sdk import Record, EncodingQuality**
ImportError: cannot import name 'EncodingQuality' from 'depthai_sdk' (/home/flip/GitHub/depthai/venv/lib/python3.11/site-packages/depthai_sdk/__init__.py)

**Expected behavior**

Clean launch.

**Screenshots**
na

**Pipeline Graph**
na

**Attach system log**
na

**Additional context**
This bug appears in a different repo, https://github.com/luxonis/depthai-experiments/issues/478 . It appears there is some merge problem where `EncodingQuality` got dropped. It doesn't exist in the API documentation, and doesn't appear in the `develop` branch either.

MartinMotycka, 2024-04-22:
@themarpe  Please check this. Thank you.

themarpe, 2024-04-22:
^ @asahtik

jtotzpr, 2025-01-21:
Same problem happens in this repo as well: https://github.com/luxonis/depthai/blob/main/apps/record/main.py#L13