luxonis/depthai #1163: [BUG] Get collapsed ROSBAG file when running ROSBAG Recording code sample

DungMinhDao, 2024-03-25:
**Describe the bug**
The file generated from the [ROSBAG Recording](https://docs.luxonis.com/projects/sdk/en/latest/samples/recording/SDK_rosbag_record/) code sample—specifically, the `recording.bag` file—is corrupted and cannot be opened with either Foxglove Studio or using  `bagpy` library to read

**Minimal Reproducible Example**
Device: Oak-D Pro Auto-focus
OS: Windows 11
Environment:
```
depthai                   2.22.0.0                 pypi_0    pypi
depthai-pipeline-graph    0.0.5                    pypi_0    pypi
depthai-sdk               1.13.1                   pypi_0    pypi
```
I use the code for `record.py` from the source code: [rosbag_record.py](https://github.com/luxonis/depthai/blob/main/depthai_sdk/examples/recording/rosbag_record.py) mentioned in the code sample 

The package `rosbags` is also installed through `pip`

**Expected behavior**
When I ran the command `python record.py` and press Ctrl+C to stop, it output the video in the directory `record/4-19443010C1384E1300/` which contains 2 files `calib.json` and `recording.bag`
```
(base) PS C:\Users\dminhdao\ABCD\depthai-experiments\rosbag> python record.py
[2024-03-25 12:32:05] INFO [root.__init__:118] Setting IR laser dot projector brightness to 800mA
[2024-03-25 12:32:06] WARNING [libav.swscaler.decode:108] deprecated pixel format used, make sure you did set range correctly
[2024-03-25 12:32:49] INFO [root.close:456] Closing OAK camera
[2024-03-25 12:32:49] INFO [root.close:418] ROS .bag saved at: C:\Users\dminhdao\ABCD\depthai-experiments\rosbag\record\4-19443010C1384E1300\recording.bag
[2024-03-25 12:32:49] INFO [root._run:33] Exiting store frame thread
[2024-03-25 12:32:50] INFO [root.close:456] Closing OAK camera
```

The file `recording.bag` has some information can be read by the `bagpy` package but its message cannot be decoded
```
$ python
Python 3.10.14 | packaged by Anaconda, Inc. | (main, Mar 21 2024, 16:20:14) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from bagpy import bagreader
>>> import pandas as pd
>>> b = bagreader('record/4-19443010C1384E1300/recording.bag')
[INFO]  Successfully created the data folder record/4-19443010C1384E1300/recording.
>>> dir(b)
['__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__weakref__', 'animate_laser', 'animate_pointcloud', 'bagfile', 'clock_data', 'compressed_images', 'datafolder', 'delimiter', 'dir', 'end_time', 'filename', 'frequency', 'laser_data', 'message_by_topic', 'message_types', 'n_messages', 'odometry_data', 'plot_odometry', 'plot_std', 'plot_vel', 'plot_wrench', 'pointcloud_data', 'reader', 'start_time', 'std_data', 'topic_table', 'topic_tuple', 'topics', 'vel_data', 'verbose', 'wrench_data']
>>> b.end_time
25.1032406
>>> b.start_time
0.0060003
>>> b.topic_table
                        Topics                        Types  Message Count  Frequency
0                 /9_depth/raw            sensor_msgs/Image            752  29.980692
1  /cam_a_bitstream/compressed  sensor_msgs/CompressedImage            753  29.944737
2  /cam_b_bitstream/compressed  sensor_msgs/CompressedImage            753  29.999850
3  /cam_c_bitstream/compressed  sensor_msgs/CompressedImage            753  29.977637
4                         /imu              sensor_msgs/Imu           4090        NaN
>>> data = b.message_by_topic('/9_depth/raw')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\dminhdao\miniconda3\lib\site-packages\bagpy\bagreader.py", line 270, in message_by_topic
    for topic, msg, t in self.reader.read_messages(topics=topic, start_time=tstart, end_time=tend):
  File "C:\Users\dminhdao\miniconda3\lib\site-packages\rosbag\bag.py", line 2705, in read_messages
    yield self.seek_and_read_message_data_record((entry.chunk_pos, entry.offset), raw, return_connection_header)
  File "C:\Users\dminhdao\miniconda3\lib\site-packages\rosbag\bag.py", line 2880, in seek_and_read_message_data_record
    msg.deserialize(data)
  File "C:\Users\dminhdao\AppData\Local\Temp\genpy_jkmp3oua\tmpdg7371ou.py", line 128, in deserialize
    self.header.frame_id = str[start:end].decode('utf-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xd0 in position 5: invalid continuation byte
```

Also when I tried to open the file with Foxglove Studio it also gives a message about corrupted file: `Failed to deserialize message on topic /9_depth/raw. RangeError: String deserialization error: length 364939000, maxLength 1843234`

**Screenshots**
This is the Foxglove Studio screenshot:
[image]

**Additional context**
The same issue persisted with that code on other Windows 11 laptop, and a more collapsed file is produced with that code on Ubuntu 22.04 with an Oak-D Pro Fixed-focus camera - the file is without the begin and end time.

LaiHoangLe, 2024-03-25:
Hi, can you provide information about the size of files?

DungMinhDao, 2024-03-25:
> Hi, can you provide information about the size of files?

Sure, the file size is 1.9 GB and recorded for about 25 seconds

cod3monk3y, 2024-03-29:
I'm seeing corrupted files with recording as well. 

Python: 3.10.10
Device: OAK-D-W

Here's a close to minimal repro:

```
def _record(record_path: Path, record_type: depthai_sdk.RecordType):
    with OakCamera() as oak:
        # for vis and Q to quit only
        left = oak.create_camera('left', resolution='800p')

        # simple recording
        imu = oak.create_imu()
        imu.config_imu(report_rate=100, batch_report_threshold=5)

        str_path = record_path.absolute()
        print(f"Recording to: {str_path}")

        #oak.record([left, right, imu], path=str_path, record_type=record_type)
        oak.record([imu], path=str_path, record_type=record_type)

        # show recording, press Q key to quit
        oak.visualize([left,])
        oak.start(blocking=True)
```

From pip:

```
depthai==2.24.0.0
depthai-pipeline-graph==0.0.5
depthai_sdk (note: install from mainline with `-e`)
depthai-viewer==0.1.8
mcap==1.1.1                                                                                                                       
mcap-ros1-support==0.7.1                                                                                                          
mcap-ros2-support==0.5.3                                                                                                          
rosbags==0.9.22                                                                                                                   
...
```

1. Run `_record` python script
2. In video window, press Q after 3-5 seconds
3. Open file in https://foxglove.dev
    1. Open local file
    2. Select *Topics* tab
    3. Verify `/imu` is the only topic
    4. Press Play button (bottom center of window)
    5. Select *Problems* tab
    6. See error message

Expected: no errors

Actual: Error message: 

```
Failed to deserialize message on topic /imu. RangeError: String deserialization error: length 781286000, maxLength 303
Check that your input file is not corrupted
```

Input file size is 139KB

Similar crash in Matlab when trying to `readMessages` on the rosbag.

I found one issue where recorder was writing `True` instead of `1` for `latching`, should be:

depthai_sdk/src/depthai_sdk/recorders/rosbag_recorder.py (line 532)
```
stream.connection = self.writer.add_connection(stream.topic, stream.ros_type.__msgtype__, latching=1)
```

But there are still other issues it seems, corrupting the recording file.

MartinMotycka, 2024-04-22:
@themarpe Please check this. Thank you.

themarpe, 2024-04-22:
^ @jakaskerl do you mind checking it out

jakaskerl, 2024-04-23:
Hi @cod3monk3y, @DungMinhDao 
Can confirm that it works on MacOS with the following dependencies: 
```
depthai-sdk                  1.12.1
depthai                         2.22.0.0
rosbags                        0.9.11
mcap                            1.1.1
mcap-ros1-support     0.0.8
```