luxonis/depthai #1168: Not able to install depthai on MacOS Monterey v12.3.1: installer exited with status 1 at line 239

abdedixit, 2024-04-09:
Hi,

I bumped into a similar issue as this one:
https://github.com/luxonis/depthai/issues/1159

Should I try switching to python v3.11.8, or is there any alternate workaround?

Please see the output from my terminal window below:

% bash -c "$(curl -fL https://docs.luxonis.com/install_depthai.sh)"
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  7975  100  7975    0     0  10747      0 --:--:-- --:--:-- --:--:-- 10747
mkdir: /Users/<username>/Luxonis: File exists

ENTER absolute installation path for depthai or leave empty and default path: $HOME will be used.

Using default installation path: /Users/<username>/Luxonis
Directory: /Users/<username>/Luxonis is OK

Python version: Python 3.12.2 found.
If you want to use it for installation, press ENTER key, otherwise input path to python binary.
Press ENTER key to continue

_____________________________
Calling macOS_installer.sh
_____________________________
Running macOS installer.
Installing global dependencies.
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  8131  100  8131    0     0  11615      0 --:--:-- --:--:-- --:--:-- 11615
bash: line 133: trixie/sid: division by 0 (error token is "d")

Failed installing dependencies. Could be a bug in the installer or unsupported platform. Open a bug report over at https://github.com/luxonis/depthai - exited with status 1 at line 239

MartinMotycka, 2024-04-22:
@themarpe Please check this. Thank you.

themarpe, 2024-04-22:
@PetrNovota any insights here?

Mnlohani, 2024-05-15:
Hi, 
I have similar issue with Mac os 12.7.5  I tried with different python version also (3.11 and 3.7(in which it automatically selected 3.10)) but didn't work. Please assist. Thanks

(base) ➜  ~ bash -c "$(curl -fL https://docs.luxonis.com/install_depthai.sh)"

% Total    % Received % Xferd  Average Speed   Time    Time     Time  Current Dload  Upload   Total   Spent    Left  Speed
0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  7975  100  7975    0     0   8457      0 --:--:-- --:--:-- --:--:--  8457
mkdir: /Users/mnl/Luxonis: File exists
ENTER absolute installation path for depthai or leave empty and default path: $HOME will be used.
Using default installation path: /Users/mnl/Luxonis
Directory: /Users/mnl/Luxonis is OK
Python version: Python 3.12.2 found.
If you want to use it for installation, press ENTER key, otherwise input path to python binary.
Press ENTER key to continue
Calling macOS_installer.sh
Running macOS installer.
Installing global dependencies.
% Total    % Received % Xferd  Average Speed   Time    Time     Time  Current Dload  Upload   Total   Spent    Left  Speed
 0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  8131  100  8131    0     0   9940      0 --:--:-- --:--:-- --:--:--  9940

bash: line 133: trixie/sid: division by 0 (error token is "d")

Failed installing dependencies. Could be a bug in the installer or unsupported platform. Open a bug report over at 
luxonis/depthai - exited with status 1 at line 239

PetrNovota, 2024-05-22:
@Mnlohani please can you run [this scrtip](https://github.com/luxonis/depthai-python/blob/main/docs/source/_static/install_dependencies.sh)
and let me know if it goes through or fails?
thank you.

abdedixit, 2024-05-30:
Hi @PetrNovota Thanks for the script. It runs fine on my Mac, and it seems to install Homebrew and git (which were already installed). See output after executing the script:

% sh TestScript.sh 
TestScript.sh: line 134: trixie/sid: division by 0 (error token is "d")
During Homebrew install, certain commands need 'sudo'. Requesting access...
Password:

 >> Installing Homebrew from https://raw.githubusercontent.com/Homebrew/install/master/install.sh
Warning: Running in non-interactive mode because `$CI` is set.
==> Checking for `sudo` access (which may request your password)...
==> This script will install:
/usr/local/bin/brew
/usr/local/share/doc/homebrew
/usr/local/share/man/man1/brew.1
/usr/local/share/zsh/site-functions/_brew
/usr/local/etc/bash_completion.d/brew
/usr/local/Homebrew
==> /usr/bin/sudo /usr/sbin/chown -R 40282643:admin /usr/local/Homebrew
==> Downloading and installing Homebrew...
remote: Enumerating objects: 3, done.
remote: Counting objects: 100% (3/3), done.
remote: Compressing objects: 100% (3/3), done.
remote: Total 3 (delta 0), reused 0 (delta 0), pack-reused 0
Unpacking objects: 100% (3/3), 2.47 KiB | 1.23 MiB/s, done.
From https://github.com/Homebrew/brew
   443e4e5dac..cb3101c8b7  master     -> origin/master
Reset branch 'stable'
==> Updating Homebrew...
==> Installation successful!

==> Homebrew has enabled anonymous aggregate formulae and cask analytics.
Read the analytics documentation (and how to opt-out) here:
  https://docs.brew.sh/Analytics
No analytics data has been sent yet (nor will any be during this install run).

==> Homebrew is run entirely by unpaid volunteers. Please consider donating:
  https://github.com/Homebrew/brew#donations

==> Next steps:
- Run these two commands in your terminal to add Homebrew to your PATH:
    (echo; echo 'eval "$(/usr/local/bin/brew shellenv)"') >> /Users/40282643/.zprofile
    eval "$(/usr/local/bin/brew shellenv)"
- Run brew help to get started
- Further documentation:
    https://docs.brew.sh

 >> brew install git
Warning: git 2.45.1 is already installed and up-to-date.
To reinstall 2.45.1, run:
  brew reinstall git

=== Installed successfully!  IMPORTANT: For changes to take effect,
please close and reopen the terminal window, or run:  exec $SHELL
Finished installing global libraries.

However, I then tried running Depth AI viewer by first quitting and reopening the terminal, and then running:
~ % depthai-viewer
zsh: command not found: depthai-viewer

I also tried running:
~ % python3 -m depthai_viewer
/usr/local/opt/python@3.12/bin/python3.12: No module named depthai_viewer

Please let me know how to proceed with the installation of depthai viewer.

Thank you.

med-tim, 2024-06-26:
experiencing this same issue with ios 14.5 and python 3.10.

Has anyone found a fix for this?

studiohuahong, 2024-07-09:
Same issue here

Mnlohani, 2024-07-10:
Hi, 
sry @PetrNovota  I missed the email and forgot to  comment here.  I posted in forum  and I got the below answer which worked for me.

https://discuss.luxonis.com/d/4759-installation-issue-mac
"""
Please use the manual approach, since the script has to be patched.

Just run curl -fL https://docs.luxonis.com/install_dependencies.sh | bash, then clone the [depthai-python](https://github.com/luxonis/depthai-python) repo and run /examples/install_requirements.py
"""
Hope It will be a fix for others also.

scott-robotics, 2024-08-22:
macOS runs bash 3.2 and can't handle associative arrays (dictionaries):

``` bash
declare -A debian_versions=(
  ["trixie/sid"]="13"
  ["bookworm/sid"]="12"
  ["bullseye/sid"]="11"
  ["buster/sid"]="10"
  ["stretch/sid"]="9"
  ["jessie/sid"]="8"
  ["wheezy/sid"]="7"
  ["squeeze/sid"]="6"
)
```

Not sure of an elegant alternative.
Or, `brew install bash` and symlink to the homebrew version

PetrNovota, 2024-08-23:
Bash 4.0 or higher is needed on macOS for the installer to work correctly. Script now checks the bash version and terminates with explanation in case the bash version is lower then required

closed 2024-08-23 by PetrNovota