luxonis/depthai #1169: [BUG] {setAutoExposureLimit is not limiting exposure on OAK-1 W PoE}

matkok, 2024-04-25:
```
#!/usr/bin/env python3
import time
import numpy as np
import cv2
import depthai as dai
import threading
import queue
import logging

class Camera:
    def __init__(self):
        self.controlIn = None
        self.name = None
        self.frame = None
        self.thread = None
        self.xout_video = None
        self.videoEncoder = None
        self.qRgb = None
        self.cam = None
        self.running = False
        self.q_list = None
        self.image = None
        self.videoIn = None
        self.xoutVideo = None
        self.cam = None
        self.dai_device = None
        self.ctrl = None
        self.video = None
        self.device = None
        self.pipeline = None
        self.control = None
        self.controlQueue = None
        self.retriving_image = False
        self.connected = False
        self.last_image_time = time.time()  # Track the time of the last image retrieval
        self.timeout_for_disconected = 5  # Track the time of the last image retrieval
        self.timeout_for_last_image = 1  # Set a timeout duration in seconds

    def create_rgb_cam_pipeline(self):
        print("Creating pipeline: RGB CAM -> XLINK OUT")
        self.pipeline = dai.Pipeline()

        self.cam = self.pipeline.create(dai.node.ColorCamera)        
        # self.xout_preview = self.pipeline.create(dai.node.XLinkOut)
        self.xout_video = self.pipeline.create(dai.node.XLinkOut)
        self.videoEncoder = self.pipeline.createVideoEncoder()

        self.ctrl = dai.CameraControl()

        self.pipeline.setXLinkChunkSize(0)

        self.xout_video.setStreamName("video")

        self.cam.setResolution(dai.ColorCameraProperties.SensorResolution.THE_4_K)
        self.cam.setInterleaved(False)
        self.cam.setBoardSocket(dai.CameraBoardSocket.CAM_A)

        self.cam.initialControl.setAutoExposureLimit(maxExposureTimeUs=5000)  ##############################################      

        self.cam.setFps(15)
        self.cam.setIsp3aFps(15)

        self.controlIn = self.pipeline.create(dai.node.XLinkIn)

        self.videoEncoder.setDefaultProfilePreset(self.cam.getVideoSize(), self.cam.getFps(), dai.VideoEncoderProperties.Profile.MJPEG)

        self.xout_video.input.setBlocking(False)
        self.xout_video.input.setQueueSize(1)
        
        self.controlIn.setStreamName('control')

        self.cam.video.link(self.videoEncoder.input)
        self.videoEncoder.bitstream.link(self.xout_video.input)
        self.controlIn.out.link(self.cam.inputControl)

        return self.pipeline

    def grab_image(self):
        return self.frame

    def get_camera_resolution(self):
        return self.self.cam.getIspSize()
    
    def set_camera_settings(self):
        self.ctrl.setAutoExposureLimit(maxExposureTimeUs=5000) ##############################################
        self.controlQueue.send(self.ctrl)
        pass

    def open(self):        
        stopwatch = time.time_ns()
        self.thread = threading.Thread(target=self.start_pipeline, daemon=True)    
        self.thread.start()
        elapsed_sec = (time.time_ns() - stopwatch) / 1e9
        print("Time to initialize camera: ", elapsed_sec)

    def start_pipeline(self):
        end_time = time.time()
        print("Creating DepthAI device")
        self.device = None
        cv2.namedWindow("Resized Image", cv2.WINDOW_NORMAL)
        cv2.resizeWindow("Resized Image", 640, 480)        

        while True:
            if self.device is None:
                try:     
                    pipeline = self.create_rgb_cam_pipeline()               
                    with dai.Device(pipeline) as self.device:                        
                        
                        print("Starting pipeline")

                        self.controlQueue = self.device.getInputQueue('control')

                        self.qRgb = self.device.getOutputQueue(name="video", maxSize=1, blocking=False)
                    
                        self.running = True
                        self.connected = True                        

                        while self.running:
                            self.set_camera_settings()
                            inRgbs = self.qRgb.tryGetAll()  # blocking call, will wait until a new data has arrived

                            for inRgb in inRgbs:
                                self.frame = cv2.imdecode(inRgb.getData(), cv2.IMREAD_UNCHANGED)
                                print(f"Exposure: {inRgb.getExposureTime().total_seconds()*1000:.3f} ms, ")
                                #print(f"ISO: {inRgb.getSensitivity()}, ")
                                cv2.imshow("Resized Image", self.frame)
                                end_time = time.time()
                                if cv2.waitKey(1) == ord('q'):
                                    break
                            time.sleep(0.3)
                except Exception as e:
                    logging.error("Exception occurred in start_pipeline", exc_info=True)
                    self.device = None
                    self.frame = None
                    self.connected = False
                    self.running = False
            else:
                logging.info("Device already initialized, waiting for the next iteration")
                self.device = None
                self.frame = None
                self.connected = False
                self.running = False

    @staticmethod
    def test():
        camera = Camera()
        camera.open()
        while(1):
            pass

if __name__ == "__main__":
    Camera.test()
```

Hi, I have just bought OAK-1 W PoE camera that I tried to use but setting setAutoExposureLimit in initial control and in control mode is not limiting my exposure. This is the output of the running code:

> Starting pipeline

> Exposure: 29.995 ms,
> 
> Exposure: 4.996 ms,
> 
> Exposure: 4.996 ms,
> 
> Exposure: 4.996 ms,
> 
> Exposure: 29.995 ms,
> 
> Exposure: 29.995 ms,
> 
> Exposure: 29.995 ms,
> 
> Exposure: 29.995 ms,
> 
> Exposure: 29.995 ms,
> 
> Exposure: 29.995 ms,
> 
> Exposure: 29.995 ms,
> 
> Exposure: 29.995 ms,
> 
> Exposure: 4.996 ms,
> 
> Exposure: 4.996 ms,

Do you have any idea what the problem might be?

Thank you in advance 🙂

MartinMotycka, 2024-05-09:
@themarpe ^^

alex-luxonis, 2024-05-13:
@matkok Can you try updating to the latest depthai, currently `2.25.1.0`.
I believe it's an issue that we fixed starting with [2.25.0.0](https://github.com/luxonis/depthai-python/releases/tag/v2.25.0.0).

matkok, 2024-05-27:
yes, that solved my problem. Thank you!
(actually I was starting my app with python and not python3 where the last version was installed).