luxonis/depthai #1171: [BUG] SDK - Replay occasionally takes depth recording for color camera

RiccardoGrieco, 2024-04-30:
**Describe the bug**

When replaying recorded data, occasionally  `6_depth`  is used for `color` camera.

**Minimal Reproducible Example**

Recording script:
```python
from depthai_sdk import OakCamera
from depthai import Device, ColorCameraProperties, MonoCameraProperties
fps = 30.0
record_dir="/tmp/oak_recording/24-04-30-15_08_17/1-1844301011A3860E00/1-1844301011A3860E00"
with OakCamera() as oak:
        # create streams
        color = oak.camera('color', resolution=ColorCameraProperties.SensorResolution.THE_1080_P, fps=fps, encode=None)
        stereo = oak.stereo(resolution=MonoCameraProperties.SensorResolution.THE_720_P, fps=fps, encode=None)
        stereo.config_stereo(align=color)

        # configure recording
        oak.record([color, stereo, stereo.left, stereo.right], path=record_dir)

        # start recording. Use CTRL+C to stop.
        oak.start(blocking=True)
```

Replay script:
```python
from depthai_sdk import OakCamera
import depthai as dai

with OakCamera(replay='/tmp/oak_recording/24-04-30-15_08_17/1-1844301011A3860E00/1-1844301011A3860E00') as oak:
    oak.replay.set_loop(True)  # <--- Enable looping of the video, so it will never end

    color = oak.create_camera('color')
    
    oak.visualize([color], fps=True)
    oak.start(blocking=True)
```

**Expected behavior**

`CAM_A_video` is used for `color` camera.

**Screenshots**

Here a screenshot of the replay script and the visualization:

[image]

**Additional context**

Library versions:
```
depthai                   2.25.0.0
depthai-pipeline-graph    0.0.5
depthai-sdk               1.13.1
```

I dig a bit into the issue and looks like the problem lies in this piece of code, where `6_depth` is selected as stream name:

https://github.com/luxonis/depthai/blob/f4a0d3d4364565faacf3ce9f131a42b2b951ec1b/depthai_sdk/src/depthai_sdk/components/camera_component.py#L78-L86

Reason behind this is that when streams are initialized, `6_depth` is assigned a `None` socket:

https://github.com/luxonis/depthai/blob/f4a0d3d4364565faacf3ce9f131a42b2b951ec1b/depthai_sdk/src/depthai_sdk/readers/videocap_reader.py#L39-L67

And if socket is `None` in `ReplayStream`, method `get_socket` returns by default CAM_A socket:

https://github.com/luxonis/depthai/blob/f4a0d3d4364565faacf3ce9f131a42b2b951ec1b/depthai_sdk/src/depthai_sdk/replay.py#L50-L59

>:warning:
NOTE that this issue is strictly dependent on the order of streams returned by `self._replay.streams.items()`. I tried with two different PCs and one has the correct behaviour, because `CAM_A_video` is read first.

MartinMotycka, 2024-05-20:
@themarpe Check this, please.

themarpe, 2024-05-21:
@Erol444 do you intend we check this behavior out?

Erol444, 2024-05-21:
Hi @RiccardoGrieco ,
As we're working on depthai v3 (which will supersede depthai-sdk), we, unfortunately, won't be focusing on SDK anymore. More info here:
https://github.com/luxonis/depthai-python/tree/v3_develop
Kind regards, Erik

closed 2024-05-21 by Erol444