luxonis/depthai #1173: [BUG] Enabling Dot projector (Active sensing) on ROS2

jaggiK, 2024-05-03:
**Describe the bug**

Not able to enable dot projector on OAK-D PRO W via ROS2. This is a follow up from [this](https://discuss.luxonis.com/d/3898-active-stereo-mode-on-ros2) thread.

Here is the camera.yaml settings to enable dot projector: `i_enable_ir: true` and `i_laser_dot_brightness: 800`. 

Launching `camera.launch.py`, `rgbd_pcl.launch.py`, or `pointcloud.launch.py` files using `ros2 launch depthai_ros_driver <launch_file>` doesn't produce point cloud data. `ros2 topic list` shows `/oak/points` topic. However, `ros2 topic echo /oak/points` doesn't print any data or rviz2 doesn't show point cloud data. I suspect that the point cloud data is not being published.

As an alternate experiment, `stereo_inertial_node.launch.py` was modified to set ` enableDotProjector` to `true` and `dotProjectormA` to `800.0`. While launching this file did produce point cloud data published on `/stereo/points` topic, the point cloud data quality wasn't better than default setting without enabling dot projector. This is tested in both indoor and outdoor setting. A sample output is attached below. 

**Minimal Reproducible Example**
OS: Ubuntu 20.04

```
depthai                              2.25.1.0
depthai-pipeline-graph               0.0.5
depthai-sdk                          1.13.1
depthai-viewer                       0.1.4
```

ROS2 version: Foxy

**Expected behavior**
Able to view high quality point cloud on rviz2 using active sensing mode.

**Screenshot**
[image]

A few questions:

- What would be the best example to enable dot projector and visualize point cloud on rviz2?
- Is there a way to confirm that dot projector is enabled?
- Like in depthai sdk directly how visualization of IR frames is possible, is it possible to view these IR frames via ROS2 interface?.
- In the screenshot above obtained from `stereo_inertial_node.launch.py`, the region marked with white contains straight line patterns. Is this expected in active sensing mode?

Thank you.

MartinMotycka, 2024-05-20:
@themarpe Please, check this.

themarpe, 2024-05-21:
@Serafadam any thoughts (also, perhaps lets move this issue to depthai-ros repository)

Serafadam, 2024-05-21:
Hi,
Unfortunately Foxy distribution is no longer supported as it reached end-of-life, I would suggest trying out Humble if possible (you can also use it inside Docker container).

> What would be the best example to enable dot projector and visualize point cloud on rviz2?
All launch file examples in depthai_ros_driver have dot projector active if present on the device.

> Is there a way to confirm that dot projector is enabled?
If you check left/right image from stereo pair, dots should be visible.

> Like in depthai sdk directly how visualization of IR frames is possible, is it possible to view these IR frames via ROS2 interface?.
These can be enabled in YAML file for depthai_ros_driver.

> In the screenshot above obtained from stereo_inertial_node.launch.py, the region marked with white contains straight line patterns. Is this expected in active sensing mode?

These probably come from aligning non-rectified rgb image in stereo_inertial_node. This node serves more as an example how to write your own driver instead of being full-fledged driver.

Serafadam, 2024-06-10:
Closing due to inactivity, please reopen if the issue still exists.

closed 2024-06-10 by Serafadam