luxonis/depthai #1177: Camera RuntimeError: Failed to boot device! - [X_LINK_UNBOOTED]

jprodriguezg, 2024-05-31:
Hello there, 

I am having issues by starting the OAK-D PRO WIDE. After using the camera (e.g., running `depthai_demo.py`), stopping the process and starting again the camera remains UNBOOTED (sometimes).

**Check if issue already exists**

There are similar errors but not exactly the same https://github.com/luxonis/depthai/issues/753 ,https://github.com/luxonis/depthai/issues/886 , https://github.com/luxonis/depthai-ros-examples/issues/57 , https://discuss.luxonis.com/d/1172-device-recognized-but-will-not-boot

**Describe the bug**
When trying to re-open the camera (e.g., running `depthai_demo.py`), I sometimes got the following error:

```
[0] 19443010B1C7992E00 [X_LINK_UNBOOTED]
  File "/home/XXX/Software/luxonis/depthai-main/depthai_demo.py", line 657, in run
    self.instance.run_all(self.conf)
  File "/home/XXX/Software/luxonis/depthai-main/depthai_demo.py", line 132, in run_all
    self.setup(conf)
  File "/home/XXX/Software/luxonis/depthai-main/depthai_demo.py", line 210, in setup
    self._device = da.Device(self._pm.pipeline.getOpenVINOVersion(), self._deviceInfo, maxUsbSpeed)
RuntimeError: Failed to boot device!
```
**Hardware**
OAK-D PRO WIDE

**Host OS (clean installation)**
[Pop!_OS 22.04](https://www.omgubuntu.co.uk/2022/04/pop-os-22-04-lts-features-download) 

**Minimal Reproducible Example**

1. [Install DepthAI](https://github.com/luxonis/depthai?tab=readme-ov-file#installation) SDK
2. Try `depthai_demo.py` several times or any other [script](https://docs.luxonis.com/projects/sdk/en/latest/quickstart/#creating-color-and-depth-streams) that starts the camera 

**Expected behavior**
I am testing the camera and trying to get  images from the device. For instance, by running the demo I expect to visualize the demo.

**Screenshots**
[image]

**Additional comments**
So far I have tried these following:

- [Setting Udev rules](https://docs.luxonis.com/en/latest/pages/troubleshooting/#udev-rules-on-linux) 
- [Forcing USB2](https://docs.luxonis.com/en/latest/pages/troubleshooting/#forcing-usb2-communication)
- Changing UBS-3 cable
- When I run `lsusb | grep 03e7` I get  `Bus 001 Device 096: ID 03e7:2485 Intel Movidius MyriadX`
- dmesg -w 
```
[85649.191911] usb 1-2: new high-speed USB device number 97 using xhci_hcd
[85649.340889] usb 1-2: New USB device found, idVendor=03e7, idProduct=2485, bcdDevice= 0.01
[85649.340895] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[85649.340896] usb 1-2: Product: Movidius MyriadX
[85649.340898] usb 1-2: Manufacturer: Movidius Ltd.
```

**Temporal solution**
So far the only solution that works is to reboot the host pc. However, the solution is not suitable for my application as I should not reboot anytime I need to use the camera.

MartinMotycka, 2024-06-24:
@themarpe ^^

themarpe, 2024-06-24:
@jprodriguezg thanks for the report - have you tested a different host as well? Some hosts have odd USB controllers onboard, that causes issues with OAK devices.
Do you mind letting us know what system is this reproducible on, for us to give it a go?