luxonis/depthai #1180: Seeking Solution to Get Color Rectified Images with OAK-FFC 4P and OV9782 Cameras

tarekmuallim, 2024-06-07:
Hi,

I am using an OAK-FFC 4P board with two OV9782 color camera modules. I want to obtain rectified left and right images, but the stereo node only provides grayscale rectified images. Is there a way to get rectified color images, or is there a workaround to calculate them?

artur-ag, 2024-09-04:
I am also interested in this. I found [this thread](https://discuss.luxonis.com/d/1526-get-rectified-color-images-from-stereocomponent/6), but I'm finding that the output of the ImageManip node is not exactly the same as the stereo's rectified output...

tarekmuallim, 2024-09-04:
Hi @artur-ag 

I tried the example provided at [this link](https://github.com/luxonis/depthai-experiments/tree/master/gen2-stereo-on-host), and it works well. I modified the rectification function to handle three channels instead of just one.

However, I'm looking for a solution to obtain the rectified color frame directly from the OAK module itself.

artur-ag, 2024-09-05:
@tarekmuallim I believe I fixed the problem on the code of the other thread ([see PR](https://github.com/luxonis/depthai-python/pull/1069)). With this, I am now getting rectified RGB from the OAK module (from ImageManip) that looks the same as the rectified grayscale coming from the StereoDepth node. However, I had to change `meshCellSize = 16` to `20`, otherwise I get a glitchy image in blocks.