luxonis/depthai #1183: [BUG] {Unsupported resolution set for detected camera IMX378/214, needs 1080_P / 4_K / 12_MP. Defaulting to 1080_P}

onthegrid007, 2024-06-29:
Hi, I am trying to get more fps out of my center rgb camera on my OAK-D by lowering the resolution but keep getting this error... Is there any way to bypass it?

onthegrid007, 2024-06-30:
I thought the side depth cameras can reach up to 120fps

themarpe, 2024-06-30:
These are the only supported resolutions for the center camera @onthegrid007 

LEFT & RIGHT ov9282 should work however at higher FPS, likely at 400p resolution CC: @alex-luxonis

closed 2024-06-30 by themarpe

alex-luxonis, 2024-07-01:
Yes, the OV9282 cameras should work at higher FPS, some info here: https://docs.luxonis.com/hardware/sensors/OV9282
but actually slightly more is possible as in the output from `device.getConnectedCameraFeatures()`:
https://github.com/luxonis/depthai-python/blob/main/docs/source/components/nodes/color_camera.rst#full-fov

> ```
> {width: 1280, height: 720, minFps: 1.687, maxFps: 143.1, type: MONO, fov: {x:0, y: 40, width: 1280, height: 720}}
> {width: 1280, height: 800, minFps: 1.687, maxFps: 129.6, type: MONO, fov: {x:0, y: 0, width: 1280, height: 800}}
> {width: 640, height: 400, minFps: 1.687, maxFps: 255.7, type: MONO, fov: {x:0, y: 0, width: 1280, height: 800}}
> ```

For IMX378 rgb camera, up to 85fps is possible now with `THE_2024x1520`: https://docs.luxonis.com/hardware/sensors/IMX378

We'll add back binning as an option for IMX378, then up to 118fps will be possible with `THE_1080_P` https://github.com/luxonis/depthai-python/pull/986 https://github.com/luxonis/depthai-core/pull/972