luxonis/depthai #1185: [BUG] module compiled against ABI version 0x1000009 but this version of numpy is 0x2000000

flabrosse, 2024-07-08:
Ubuntu 22.04
depthai 2.27.0.0

I just installed depthai following the instructions [here](https://docs.luxonis.com/hardware/platform/calibration) (because I need to calibrate my camera) and [here](https://github.com/luxonis/depthai).

None of the python scripts work:
```
$ python3 depthai_demo.py 
RuntimeError: module compiled against ABI version 0x1000009 but this version of numpy is 0x2000000
Third party libraries failed to import: numpy.core.multiarray failed to import
Run "python3 install_requirements.py" to install dependencies
```
and
```
$ python3 calibrate.py 
RuntimeError: module compiled against ABI version 0x1000009 but this version of numpy is 0x2000000
Traceback (most recent call last):
  File "/home/ffl/local/src/depthai/calibrate.py", line 17, in <module>
    import cv2
  File "/home/ffl/.local/lib/python3.10/site-packages/cv2/__init__.py", line 8, in <module>
    from .cv2 import *
ImportError: numpy.core.multiarray failed to import
```
I have seen this mentioned here and there, and it having been fixed in depthai 2.27 but:
```
$ depthai -v
2.27.0.0
```

themarpe, 2024-07-08:
@flabrosse try downgreading the numpy for the time being. I think that opencv also hasn't come up with new wheels just yet.

@jakaskerl do you mind fixing this in the depthai_demo.py dependencies in the meantime (specifying numpy explicitly)

jakaskerl, 2024-07-09:
OpenCVs latest version 4.10.0.84 works with numpy==2.0.0, the others don't. I'll update the reqs.

flabrosse, 2024-07-09:
It looks like ubuntu 22.04 only had opencv-4.5.

flabrosse, 2024-07-09:
The version of numpy installed is 1.21.5:
```
python3-numpy is already the newest version (1:1.21.5-1ubuntu22.04.1).
```
However:
```
$ python3 -c "import numpy; print(numpy.__version__)"
2.0.0
```
It's looking like numpy 2 was installed by one of your dependencies as I now have numpy 2 in my account in site-packages.  If I move them out of the way:
```
$ python3 -c "import numpy; print(numpy.__version__)"
1.21.5
```
However, if I try to run depthai_demo.py, it complains:
```
$ python3 depthai_demo.py
Using depthai module from:  /home/ffl/.local/lib/python3.10/site-packages/depthai.cpython-310-x86_64-linux-gnu.so
Depthai version installed:  2.27.0.0
Version mismatch between installed depthai lib and the required one by the script. 
                Required:  2.24.0.0
                Installed: 2.27.0.0
                Run: python3 install_requirements.py
```
and rerunning install_requirements.py reinstalled numpy 2.

flabrosse, 2024-07-15:
I just figured out that the issue I have after removing numpy 2 is due to the fact that I also had depthai installed from a ROS package.  Removing it and the demo works (at least starts, and then fails because of lacking calibration).

Having numpy 2 around still leads to the original error.  So there is still an issue with installing numpy 2.

flabrosse, 2024-07-15:
Nothing is ever simple:
```
$ python3 calibrate.py 
calibrate.py:12: UserWarning: A NumPy version >=1.23.5 and <2.3.0 is required for this version of SciPy (detected version 1.21.5)
  from scipy.spatial.transform import Rotation
```

jakaskerl, 2024-07-15:
@flabrosse 
What machine are you running this on? I merged the requirements and it works on my side now. It should install numpy==1.26.4 which is compatible with opencv-contrib-python==4.5.5.62. Unless you are running on 32bit OS, this should work I think.

flabrosse, 2024-07-15:
I had done a pull earlier today and there was no change.  A new pull just after your message brought the change in, and now I have a calibration.  Thank you!

closed 2024-07-15 by flabrosse