luxonis/depthai #1202: [BUG] RuntimeError: module compiled against ABI version 0x1000009 but this version of numpy is 0x2000000

MRo47, 2024-09-03:
I'm trying to run the docker image 

```bash
docker run --privileged -v /dev/bus/usb:/dev/bus/usb --device-cgroup-rule='c 189:* rmw' -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix --network host --rm -i -t luxonis/depthai:latest python3 /depthai/depthai_demo.py
```

I get the following error

```bash
RuntimeError: module compiled against ABI version 0x1000009 but this version of numpy is 0x2000000
Third party libraries failed to import: numpy.core.multiarray failed to import
Run "python3 install_requirements.py" to install dependencies
```

Seems like this was seen before: #1185

JasperTan97, 2025-05-01:
Hello, I found that they're pulling numpy v2 and opencv requires numpy v1.

Overall, I managed to get the rgb preview by changing the versions of numpy and opencv. This worked for me.
```
numpy==1.26.4
opencv-python==4.8.0.74

MRo47, 2025-05-01:
Makes sense, the `numpy<2.0`, AFAIK there is no `numpy 3.0` yet
thanks  @JasperTan97