luxonis/depthai #1208: [BUG] Recording neural network output throws KeyError: '2_out;0_preview'

takgoya, 2024-10-02:
Following your documentation for Recording (https://docs-old.luxonis.com/projects/sdk/en/latest/quickstart/#recording), I have tried to record both the color camera and the neural network outputs.

Code is as simple as in your example:
```
from depthai_sdk import OakCamera
from depthai_sdk.record import RecordType

with OakCamera() as oak:
    color = oak.create_camera('color', resolution='1080p')
    yolo = oak.create_nn('yolov6n_coco_640x640', input=color)

    oak.record([color, yolo], path='./records', record_type=RecordType.VIDEO)
    oak.start(blocking=True)
```

It throws the following error:
File: "..\site-packages\depthai_sdk\recorders\video_recorder.py", line 98, in write
    self._writers[name].write(frame)
KeyError: '2_out;0_preview'

It happens with any neural network, I have tried with my own customized model, and it gives the same error.

takgoya, 2024-10-02:
Camera: OAK-D W PoE OV9782

Conda environment in Windows, with:
- Python version: 3.10.14
- depthai==2.28.0.0
- depthai-sdk==1.15.0

Without recording camera works properly, running the neural network with no problem.
Everything works fine.

takgoya, 2024-10-02:
Yep, the color channel has no problem with the recording.

The problems appears when the channel to be recorded is the one of the neural network.

takgoya, 2024-10-02:
Cool! Many thanks.
I will try it.

takgoya, 2024-10-03:
I have tried it and for me it works setting the encode to H265. With H264 it throws an error when closing the video.

Anyway, what I have seen is that the neural network recording does not contain any detection, it looks like the raw video.

I have also seen that there is some parameter that can be passed to Visualize to set a record_path, but it is not further used.

It seems that this recording feature is not developed.

Erol444, 2024-10-03:
Hi @takgoya ,
Video recorder can't record NN outputs, only raw sensor streams (camera sensors, imu..). For now, I'd suggest using callback for `yolo` output, and add cv2.videoWriter inside it.
```
oak.visualize(yolo, callback=cb_function)
```
[image]

@brmarkus , could you please contact me at erik@luxonis.com?

takgoya, 2024-10-03:
Thanks for the replies.

Yes, I have implemented a callback to call cv2.Videowriter in order to record.