luxonis/depthai #1215: [BUG] Device crashed, with dump report

qiangxinglin, 2024-10-28:
**Check if issue already exists**

- Google it (e.g. *error xy* github luxonis depthai)
- Check [troubleshooting](https://docs.luxonis.com/en/latest/pages/troubleshooting/) in documentation.

**Describe the bug**
Not sure what happened. When I press 'q' to stop the program, it generates the dump reports.

**Minimal Reproducible Example**
The crash only appeared once. And the code is quite simple.
```
import depthai as dai
import cv2
from datetime import timedelta

pipeline = dai.Pipeline()

left = pipeline.create(dai.node.ColorCamera)
right = pipeline.create(dai.node.ColorCamera)
sync = pipeline.create(dai.node.Sync)
out = pipeline.create(dai.node.XLinkOut)

left.setResolution(dai.ColorCameraProperties.SensorResolution.THE_800_P)
left.setCamera("left")
right.setResolution(dai.ColorCameraProperties.SensorResolution.THE_800_P)
right.setCamera("right")
out.setStreamName("out")
sync.setSyncThreshold(timedelta(milliseconds=30))

left.video.link(sync.inputs["left"])
right.video.link(sync.inputs["right"])
sync.out.link(out.input)

with dai.Device(pipeline) as device:
    queue = device.getOutputQueue("out", 10, False)
    while True:
        msgGrp = queue.get()
        for name, msg in msgGrp:
            frame = msg.getCvFrame()
            cv2.imshow(name, frame)
        if cv2.waitKey(1) == ord("q"):
            break
```

**Expected behavior**
The program should be quited cleanly.

**Screenshots**
```
[2024-10-28 13:35:34.244] [depthai] [error] Device with id 14442C10719237D200 has crashed. Crash dump logs are stored in: E:\OAK\.cache\depthai\crashdumps\eae46b6bad4c81652d444811160a8829a8b00f29\crash_dump.json - please report to developers.
```

**Pipeline Graph**

N/A

**Attach system log**
```
{
    "architecture": "64bit WindowsPE",
    "machine": "AMD64",
    "platform": "Windows-10-10.0.19045-SP0",
    "processor": "Intel64 Family 6 Model 167 Stepping 1, GenuineIntel",
    "python_build": "main Sep 11 2023 14:09:26",
    "python_compiler": "MSC v.1916 64 bit (AMD64)",
    "python_implementation": "CPython",
    "python_version": "3.9.18",
    "release": "10",
    "system": "Windows",
    "version": "10.0.19045",
    "win32_ver": "10 10.0.19045 SP0 Multiprocessor Free",
    "packages": [
        "aiohttp==3.9.3",
        "aiosignal==1.3.1",
        "asttokens==2.4.1",
        "async-timeout==4.0.3",
        "attrs==23.2.0",
        "blinker==1.8.2",
        "blobconverter==1.4.3",
        "certifi==2024.2.2",
        "cffi==1.16.0",
        "charset-normalizer==3.3.2",
        "click==8.1.7",
        "colorama==0.4.6",
        "coloredlogs==15.0.1",
        "comm==0.2.1",
        "ConfigArgParse==1.7",
        "contourpy==1.2.0",
        "cos-python-sdk-v5==1.9.27",
        "crcmod==1.7",
        "cycler==0.12.1",
        "Cython==3.0.11",
        "dash==2.18.1",
        "dash-core-components==2.0.0",
        "dash-html-components==2.0.0",
        "dash-table==5.0.0",
        "debugpy==1.8.1",
        "decorator==5.1.1",
        "depthai==2.28.0.0",
        "et-xmlfile==1.1.0",
        "exceptiongroup==1.2.0",
        "executing==2.0.1",
        "fastjsonschema==2.20.0",
        "Flask==3.0.3",
        "flatbuffers==24.3.25",
        "fonttools==4.49.0",
        "frozenlist==1.4.1",
        "grpc-interceptor==0.15.4",
        "grpcio==1.62.1",
        "grpcio-tools==1.62.1",
        "humanfriendly==10.0",
        "idna==3.6",
        "imageio==2.34.0",
        "importlib-metadata==7.0.1",
        "importlib_resources==6.1.2",
        "ipykernel==6.29.3",
        "ipython==8.18.1",
        "ipywidgets==8.1.5",
        "itsdangerous==2.2.0",
        "jedi==0.19.1",
        "Jinja2==3.1.4",
        "joblib==1.4.2",
        "jsonschema==4.23.0",
        "jsonschema-specifications==2024.10.1",
        "jupyter_client==8.6.0",
        "jupyter_core==5.7.1",
        "jupyterlab_widgets==3.0.13",
        "kiwisolver==1.4.5",
        "labelme2yolo==0.1.7",
        "lazy_loader==0.4",
        "ldap3==2.9.1",
        "markdown-it-py==3.0.0",
        "MarkupSafe==3.0.2",
        "matplotlib==3.8.3",
        "matplotlib-inline==0.1.6",
        "mdurl==0.1.2",
        "mpmath==1.3.0",
        "multidict==6.0.5",
        "natsort==8.4.0",
        "nbformat==5.10.4",
        "nest-asyncio==1.6.0",
        "networkx==3.2.1",
        "Nuitka==2.2.1",
        "numpy==1.26.4",
        "onnxruntime-gpu==1.17.1",
        "open3d==0.18.0",
        "opencv-python==4.9.0.80",
        "openpyxl==3.1.2",
        "openvino==2024.3.0",
        "openvino-telemetry==2023.2.1",
        "ordered-set==4.1.0",
        "packaging==23.2",
        "pandas==2.2.1",
        "parso==0.8.3",
        "pillow==10.2.0",
        "pip==23.3.1",
        "platformdirs==4.2.0",
        "plotly==5.24.1",
        "prompt-toolkit==3.0.43",
        "protobuf==4.25.3",
        "psutil==5.9.8",
        "pure-eval==0.2.2",
        "pyasn1==0.6.0",
        "PyAudio==0.2.14",
        "pycparser==2.22",
        "pycryptodome==3.20.0",
        "Pygments==2.17.2",
        "PyMuPDF==1.24.3",
        "PyMuPDFb==1.24.3",
        "PyOpenGL @ file:///C:/Program%20Files%20%28x86%29/ZED%20SDK/PyOpenGL-3.1.5-cp39-cp39-win_amd64.whl#sha256=4f6861cd8f7bd0d8b51fbd0afb10ce10102b79c620237480eaf638162bf40a1b",
        "PyOpenGL-accelerate @ file:///C:/Program%20Files%20%28x86%29/ZED%20SDK/PyOpenGL_accelerate-3.1.5-cp39-cp39-win_amd64.whl#sha256=1b9b2ccb587d1fbe39cc2f0146674b80962daf5f49bac98e1eea89e672fff8ba",
        "pyparsing==3.1.2",
        "PyQt5==5.15.10",
        "PyQt5-Qt5==5.15.2",
        "PyQt5-sip==12.13.0",
        "pyreadline3==3.4.1",
        "python-dateutil==2.9.0.post0",
        "pytz==2024.1",
        "pywin32==306",
        "PyYAML==6.0.1",
        "pyzed @ file:///C:/Program%20Files%20%28x86%29/ZED%20SDK/pyzed-4.2-cp39-cp39-win_amd64.whl#sha256=e53692af6301d14660eac363622cb712e286244600c6d35e09acad8daa9bb7e9",
        "pyzmq==25.1.2",
        "referencing==0.35.1",
        "requests==2.31.0",
        "retrying==1.3.4",
        "rich==13.7.1",
        "rpds-py==0.20.0",
        "scikit-image==0.22.0",
        "scikit-learn==1.4.2",
        "scipy==1.13.0",
        "setuptools==68.2.2",
        "six==1.16.0",
        "srt==3.5.3",
        "stack-data==0.6.3",
        "sympy==1.13.1",
        "tenacity==8.2.3",
        "textdistance==4.6.2",
        "threadpoolctl==3.5.0",
        "tifffile==2024.2.12",
        "tornado==6.4",
        "tqdm==4.66.4",
        "traitlets==5.14.1",
        "typing_extensions==4.10.0",
        "tzdata==2024.1",
        "urllib3==2.2.1",
        "vosk==0.3.45",
        "wcwidth==0.2.13",
        "websockets==12.0",
        "Werkzeug==3.0.6",
        "wheel==0.41.2",
        "widgetsnbextension==4.0.13",
        "xlrd==2.0.1",
        "xmltodict==0.13.0",
        "yarl==1.9.4",
        "zipp==3.17.0",
        "zstandard==0.22.0"
    ],
    "usb": [
        "NoLib"
    ],
    "uname": [
        "Windows DESKTOP-MV4FC1S 10 10.0.19045 AMD64 Intel64 Family 6 Model 167 Stepping 1, GenuineIntel"
    ]
}
```

**Additional context**
Device: OAK-D-SR-POE
[crash_dump.json](https://github.com/user-attachments/files/17537803/crash_dump.json)

Erol444, 2024-11-25:
Does it still throw this error @qiangxinglin ?

qiangxinglin, 2024-11-25:
No, it only happened once. I've tried so many times, but only once.

closed 2024-11-25 by Erol444