luxonis/depthai #1218: [BUG] new OAK-Thermal shows up as OAK-D Pro to demo app, fails "[error] Camera not detected on socket: 0"

MAVProxyUser, 2024-11-16:
kfinisterre@SeekOnJam0:~/depthai$ ./depthai_demo.py 
/home/kfinisterre/.local/lib/python3.10/site-packages/requests/__init__.py:102: RequestsDependencyWarning: urllib3 (2.2.3) or chardet (4.0.0)/charset_normalizer (2.0.12) doesn't match a supported version!
  warnings.warn("urllib3 ({}) or chardet ({})/charset_normalizer ({}) doesn't match a supported "
Using depthai module from:  /home/kfinisterre/.local/lib/python3.10/site-packages/depthai.cpython-310-x86_64-linux-gnu.so
Depthai version installed:  2.24.0.0
/home/kfinisterre/.local/lib/python3.10/site-packages/requests/__init__.py:102: RequestsDependencyWarning: urllib3 (2.2.3) or chardet (4.0.0)/charset_normalizer (2.0.12) doesn't match a supported version!
  warnings.warn("urllib3 ({}) or chardet ({})/charset_normalizer ({}) doesn't match a supported "
Setting up demo...
Available devices:
[0] 14442C104125EDCF00 [X_LINK_BOOTLOADER]
Disabling depth...
Disabling depth preview...
Disabling depthRaw preview...
Disabling left preview...
Disabling rectifiedLeft preview...
Disabling right preview...
Disabling rectifiedRight preview...
Enabling low-bandwidth mode due to connection mode... (protocol: XLinkProtocol.X_LINK_TCP_IP)
Setting PoE video quality to 50 to reduce latency...
IR drivers found on OAK-D Pro: ['LM3644 on bus 2']
 --> Go to the `Depth` tab to enable!
Creating MJPEG link for ColorCamera node and color xlink stream...
Downloading /home/kfinisterre/.cache/blobconverter/mobilenet-ssd_openvino_2021.4_6shave.blob...

Done
[14442C104125EDCF00] [192.168.0.100] [16.847] [ColorCamera(0)] [error] Camera not detected on socket: 0

Erol444, 2024-11-17:
Hi @MAVProxyUser , you should update your depthai version to latest one, 0.24 doesn't have support for thermal sensor yet

MAVProxyUser, 2024-11-17:
@Erol444 I'm using the version from git...

MAVProxyUser, 2024-11-17:
Standard practice dictates ./install_requirements.py which installs the version I'm using *for* me. The latest depth ai git repo automatically pulls down depthai-2.24.0.0 without my control @Erol444 

[image]

Erol444, 2024-11-17:
@MAVProxyUser depthai_demo is deprecated for some time now in favour of depthai-viewer, which has the latest version (and also support for OAK Thermal)

Erol444, 2024-11-17:
we should really remove it from this repo

MAVProxyUser, 2024-11-17:
Ok I'll give that a shot.

MAVProxyUser, 2024-11-17:
ok @Erol444 I followed the install directions here: https://github.com/luxonis/depthai-viewer 

```
kfinisterre@SeekOnJam0:~$ python3 -m pip install depthai-viewer
```

then

```
kfinisterre@SeekOnJam0:~/depthai-python/examples/Camera$ python3 -m depthai_viewer
[2024-11-17T19:46:29Z INFO  winit::platform_impl::platform::x11::window] Guessed window scale factor: 2.1666666666666665
Backend started successfully.
[2024-11-17T19:46:31Z INFO  ewebsock::native_tungstenite] WebSocket handshake has been successfully completed
[2024-11-17T19:46:31Z INFO  re_viewer::depthai::ws] Websocket opened
Selecting device:  14442C104125EDCF00
Connecting to viewer at 127.0.0.1:9876
Oak cam:  <depthai_sdk.oak_camera.OakCamera object at 0x70e1b8dd8b20>
Default stereo pair:  None
Selecting device:  14442C104125EDCF00
Resetting...
Closing device...
Done
Failed to select device: Cannot find any device with given deviceInfo
[2024-11-17T19:46:52Z ERROR re_viewer::depthai::depthai] Error: Cannot find any device with given deviceInfo, Try to connect the device to a different port.
Pipeline config:  auto=True cameras=[] stereo=None ai_model=None imu=ImuConfiguration(report_rate=100, batch_report_threshold=5)
No device selected, can't update pipeline!
[2024-11-17T19:46:53Z ERROR re_viewer::depthai::depthai] Error: No device selected, can't update pipeline!
Selecting device:  
Selecting device:  
Selecting device:  14442C104125EDCF00
Connecting to viewer at 127.0.0.1:9876
Oak cam:  <depthai_sdk.oak_camera.OakCamera object at 0x70e1b8dd8ca0>
Default stereo pair:  None
Pipeline config:  auto=True cameras=[] stereo=None ai_model=None imu=ImuConfiguration(report_rate=100, batch_report_threshold=5)
Updating pipeline...
Usb speed:  UsbSpeed.UNKNOWN
Connected to a PoE device, camera streams will be JPEG encoded...
Creating camera:  fps=30 resolution=<CameraSensorResolution.THE_256X192: 'THE_256X192'> kind=<CameraSensorType.THERMAL: 3> board_socket=<CameraBoardSocket.CAM_E: 4> stream_enabled=True name='Thermal' tof_align=None is_used_as_stereo_align=False
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/kfinisterre/.local/lib/python3.10/site-packages/depthai_viewer/_backend/main.py", line 199, in <module>
    DepthaiViewerBack(args.port, args.sdk_port)
  File "/home/kfinisterre/.local/lib/python3.10/site-packages/depthai_viewer/_backend/main.py", line 55, in __init__
    self.run()
  File "/home/kfinisterre/.local/lib/python3.10/site-packages/depthai_viewer/_backend/main.py", line 172, in run
    self.result_queue.put(self.handle_action(action, **kwargs))
  File "/home/kfinisterre/.local/lib/python3.10/site-packages/depthai_viewer/_backend/main.py", line 115, in handle_action
    message = self.on_update_pipeline(kwargs.get("runtime_only"))  # type: ignore[arg-type]
  File "/home/kfinisterre/.local/lib/python3.10/site-packages/depthai_viewer/_backend/main.py", line 101, in on_update_pipeline
    message = self._device.update_pipeline(runtime_only)
  File "/home/kfinisterre/.local/lib/python3.10/site-packages/depthai_viewer/_backend/device.py", line 422, in update_pipeline
    camera_features = next(filter(lambda feat: feat.socket == cam.board_socket, connected_camera_features))
StopIteration

```

Still no luck.

MAVProxyUser, 2024-11-17:
Assuming that the viewer installed the proper dependencies... I did not follow the instructions and stomp deps again
```
kfinisterre@SeekOnJam0:~/depthai-python/examples/ImageAlign$ python3 thermal_align.py 
Traceback (most recent call last):
  File "/home/kfinisterre/depthai-python/examples/ImageAlign/thermal_align.py", line 38, in <module>
    raise RuntimeError("No thermal camera found!")
RuntimeError: No thermal camera found!
```

MAVProxyUser, 2024-11-17:
For posterity... updated deps from the repo 
```
kfinisterre@SeekOnJam0:~/depthai-python/examples/ImageAlign$ cd ..
kfinisterre@SeekOnJam0:~/depthai-python/examples$ python3 install_requirements.py 
pip 24.3.1 from /home/kfinisterre/.local/lib/python3.10/site-packages/pip (python 3.10)
Requirement already satisfied: pip in /home/kfinisterre/.local/lib/python3.10/site-packages (24.3.1)
Requirement already satisfied: pyyaml in /home/kfinisterre/.local/lib/python3.10/site-packages (6.0.2)
Requirement already satisfied: requests in /home/kfinisterre/.local/lib/python3.10/site-packages (2.26.0)
Collecting requests
  Using cached requests-2.32.3-py3-none-any.whl.metadata (4.6 kB)
Requirement already satisfied: numpy<3.0 in /home/kfinisterre/.local/lib/python3.10/site-packages (1.26.4)
Collecting numpy<3.0
  Using cached numpy-2.1.3-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (62 kB)
Collecting opencv-python<5.0
  Using cached opencv_python-4.10.0.84-cp37-abi3-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (20 kB)
Requirement already satisfied: charset-normalizer<4,>=2 in /home/kfinisterre/.local/lib/python3.10/site-packages (from requests) (2.0.12)
Requirement already satisfied: idna<4,>=2.5 in /usr/lib/python3/dist-packages (from requests) (3.3)
Requirement already satisfied: urllib3<3,>=1.21.1 in /home/kfinisterre/.local/lib/python3.10/site-packages (from requests) (1.26.20)
Requirement already satisfied: certifi>=2017.4.17 in /usr/lib/python3/dist-packages (from requests) (2020.6.20)
Using cached requests-2.32.3-py3-none-any.whl (64 kB)
Using cached numpy-2.1.3-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (16.3 MB)
Using cached opencv_python-4.10.0.84-cp37-abi3-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (62.5 MB)
Installing collected packages: requests, numpy, opencv-python
  Attempting uninstall: requests
    Found existing installation: requests 2.26.0
    Uninstalling requests-2.26.0:
      Successfully uninstalled requests-2.26.0
  Attempting uninstall: numpy
    Found existing installation: numpy 1.26.4
    Uninstalling numpy-1.26.4:
      Successfully uninstalled numpy-1.26.4
Successfully installed numpy-2.1.3 opencv-python-4.10.0.84 requests-2.32.3
Collecting depthai
  Using cached depthai-2.28.0.0-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (8.7 kB)
Using cached depthai-2.28.0.0-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (14.0 MB)
Installing collected packages: depthai
  Attempting uninstall: depthai
    Found existing installation: depthai 2.24.0.0
    Uninstalling depthai-2.24.0.0:
      Successfully uninstalled depthai-2.24.0.0
Successfully installed depthai-2.28.0.0
################|| Downloading calibration-sample ||################

========== Retrieving /home/kfinisterre/depthai-python/examples/models/depthai_calib.json from the cache

========== Retrieving /home/kfinisterre/depthai-python/examples/models/depthai_v5.calib from the cache

========== Retrieving /home/kfinisterre/depthai-python/examples/models/BW1098OBC.json from the cache

################|| Downloading cast-models ||################

========== Retrieving /home/kfinisterre/depthai-python/examples/models/blur_simplified_openvino_2021.4_6shave.blob from the cache

========== Retrieving /home/kfinisterre/depthai-python/examples/models/diff_openvino_2022.1_6shave.blob from the cache

################|| Downloading concat-model ||################

========== Retrieving /home/kfinisterre/depthai-python/examples/models/concat_openvino_2021.4_6shave.blob from the cache

################|| Downloading mobilenet-ssd ||################

========== Retrieving /home/kfinisterre/depthai-python/examples/models/mobilenet-ssd_openvino_2021.4_6shave.blob from the cache

========== Retrieving /home/kfinisterre/depthai-python/examples/models/mobilenet-ssd_openvino_2021.4_5shave.blob from the cache

========== Retrieving /home/kfinisterre/depthai-python/examples/models/mobilenet-ssd_openvino_2021.4_8shave.blob from the cache

################|| Downloading normalization-model ||################

========== Retrieving /home/kfinisterre/depthai-python/examples/models/normalize_openvino_2021.4_4shave.blob from the cache
[... 26 lines cut ...]

========== Retrieving /home/kfinisterre/depthai-python/examples/models/construction_vest.mp4 from the cache
```

and same thing 
```
kfinisterre@SeekOnJam0:~/depthai-python/examples/ImageAlign$ python3 thermal_align.py 
Traceback (most recent call last):
  File "/home/kfinisterre/depthai-python/examples/ImageAlign/thermal_align.py", line 38, in <module>
    raise RuntimeError("No thermal camera found!")
RuntimeError: No thermal camera found!

kfinisterre@SeekOnJam0:~/depthai-python/examples/ImageAlign$ git show
commit 50f509bdd29c89817cf546bd8eb2c104f083f82f (HEAD -> main, origin/main, origin/HEAD)
Merge: 1a1eb16e 1e7f8348
Author: Erol444 <erol123444@gmail.com>
Date:   Wed Oct 23 16:36:36 2024 +0200

    Merge pull request #1079 from KenN7/patch-1
    
    Fix Camera Type check in flashconfig

kfinisterre@SeekOnJam0:~/depthai-python/examples/ImageAlign$ git branch
* main
```

Erol444, 2024-11-18:
Hi @MAVProxyUser ,
after debugging, it's very likely you'll need to use latest `develop`. Could you checkout on develop and run `install_requirements.py`, and try again? The fix will be released in 2.29 (should be released soon).

Erol444, 2024-11-18:
Here are wheels for depthai-viewer that work with latest OAK-T:
https://drive.google.com/file/d/1_jBXpGp-LQR85ikAfUTVmvQ0mpMRkHu6/view?usp=sharing

MAVProxyUser, 2024-11-18:
Thanks I'll give them a try shortly @Erol444 !

MAVProxyUser, 2024-11-18:
Looks like on Linux I get a new error @Erol444 
```
kfinisterre@SeekOnJam0:~$ depthai-viewer 
[2024-11-18T17:00:16Z INFO  winit::platform_impl::platform::x11::window] Guessed window scale factor: 1
vulkan: No DRI3 support detected - required for presentation
Note: you can probably enable DRI3 in your Xorg config
[2024-11-18T17:00:18Z WARN  egui_winit::clipboard] Failed to initialize arboard clipboard: Unknown error while interacting with the clipboard: X11 server connection timed out because it was unreachable
Got exit status: Exited(0)
Log output: Requirement already satisfied: pip in ./.local/lib/python3.10/site-packages/depthai_viewer/venv-0.2.5/lib/python3.10/site-packages (22.0.2)
Collecting pip
  Using cached pip-24.3.1-py3-none-any.whl (1.8 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 22.0.2
    Uninstalling pip-22.0.2:
      Successfully uninstalled pip-22.0.2
Successfully installed pip-24.3.1
Collecting depthai-sdk==1.14.0
  Using cached depthai_sdk-1.14.0-py3-none-any.whl.metadata (3.4 kB)
Collecting opencv-contrib-python>4 (from depthai-sdk==1.14.0)
  Using cached opencv_contrib_python-4.10.0.84-cp37-abi3-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (20 kB)
Collecting blobconverter>=1.4.1 (from depthai-sdk==1.14.0)
  Using cached blobconverter-1.4.3-py3-none-any.whl.metadata (7.8 kB)
Collecting pytube>=12.1.0 (from depthai-sdk==1.14.0)
  Using cached pytube-15.0.0-py3-none-any.whl.metadata (5.0 kB)
Collecting depthai (from depthai-sdk==1.14.0)
  Using cached depthai-2.28.0.0-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (8.7 kB)
Collecting PyTurboJPEG==1.6.4 (from depthai-sdk==1.14.0)
  Using cached PyTurboJPEG-1.6.4-py3-none-any.whl
Collecting marshmallow==3.17.0 (from depthai-sdk==1.14.0)
  Using cached marshmallow-3.17.0-py3-none-any.whl.metadata (7.8 kB)
Collecting xmltodict (from depthai-sdk==1.14.0)
  Using cached xmltodict-0.14.2-py2.py3-none-any.whl.metadata (8.0 kB)
Collecting sentry-sdk==1.21.0 (from depthai-sdk==1.14.0)
  Using cached sentry_sdk-1.21.0-py2.py3-none-any.whl.metadata (8.6 kB)
Collecting depthai-pipeline-graph==0.0.5 (from depthai-sdk==1.14.0)
  Using cached depthai_pipeline_graph-0.0.5-py3-none-any.whl.metadata (8.8 kB)
Collecting ahrs==0.3.1 (from depthai-sdk==1.14.0)
  Using cached AHRS-0.3.1-py3-none-any.whl.metadata (13 kB)
Collecting numpy>=1.21 (from depthai-sdk==1.14.0)
  Using cached numpy-2.1.3-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (62 kB)
Collecting Qt.py>=1.3.0 (from depthai-pipeline-graph==0.0.5->depthai-sdk==1.14.0)
  Using cached Qt.py-1.4.1-py2.py3-none-any.whl.metadata (29 kB)
Collecting packaging>=17.0 (from marshmallow==3.17.0->depthai-sdk==1.14.0)
  Using cached packaging-24.2-py3-none-any.whl.metadata (3.2 kB)
Collecting certifi (from sentry-sdk==1.21.0->depthai-sdk==1.14.0)
  Using cached certifi-2024.8.30-py3-none-any.whl.metadata (2.2 kB)
Collecting urllib3>=1.26.11 (from sentry-sdk==1.21.0->depthai-sdk==1.14.0)
  Using cached urllib3-2.2.3-py3-none-any.whl.metadata (6.5 kB)
Collecting requests (from blobconverter>=1.4.1->depthai-sdk==1.14.0)
  Using cached requests-2.32.3-py3-none-any.whl.metadata (4.6 kB)
Collecting PyYAML (from blobconverter>=1.4.1->depthai-sdk==1.14.0)
  Using cached PyYAML-6.0.2-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (2.1 kB)
Collecting types-PySide2 (from Qt.py>=1.3.0->depthai-pipeline-graph==0.0.5->depthai-sdk==1.14.0)
  Using cached types_pyside2-5.15.2.1.7-py2.py3-none-any.whl.metadata (8.1 kB)
Collecting charset-normalizer<4,>=2 (from requests->blobconverter>=1.4.1->depthai-sdk==1.14.0)
  Using cached charset_normalizer-3.4.0-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (34 kB)
Collecting idna<4,>=2.5 (from requests->blobconverter>=1.4.1->depthai-sdk==1.14.0)
  Using cached idna-3.10-py3-none-any.whl.metadata (10 kB)
Using cached depthai_sdk-1.14.0-py3-none-any.whl (230 kB)
Using cached AHRS-0.3.1-py3-none-any.whl (197 kB)
Using cached depthai_pipeline_graph-0.0.5-py3-none-any.whl (123 kB)
Using cached marshmallow-3.17.0-py3-none-any.whl (48 kB)
Using cached sentry_sdk-1.21.0-py2.py3-none-any.whl (199 kB)
Using cached blobconverter-1.4.3-py3-none-any.whl (10 kB)
Using cached numpy-2.1.3-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (16.3 MB)
Using cached opencv_contrib_python-4.10.0.84-cp37-abi3-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (68.7 MB)
Using cached pytube-15.0.0-py3-none-any.whl (57 kB)
Using cached depthai-2.28.0.0-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (14.0 MB)
Using cached xmltodict-0.14.2-py2.py3-none-any.whl (10.0 kB)
[... 49 lines cut ...]
      Successfully uninstalled depthai-2.28.0.0
Successfully installed deprecated-1.2.15 depthai-2.29.0.0.dev0+d3d0e8959a84ab179952aa9f97c3f62be844776c numpy-1.26.4 pyarrow-16.0.0 pydantic-1.9.0 typing-extensions-4.12.2 websockets-14.1 wrapt-1.16.0
Creating virtual environment...
Removing old venv: venv-0.2.6
Status Dump: {"venv_site_packages": "/home/kfinisterre/.local/lib/python3.10/site-packages/depthai_viewer/venv-0.2.5/lib/python3.10/site-packages"}

Backend started successfully.
[2024-11-18T17:02:19Z INFO  ewebsock::native_tungstenite] WebSocket handshake has been successfully completed
[2024-11-18T17:02:19Z INFO  re_viewer::depthai::ws] Websocket opened
Selecting device:  14442C104125EDCF00
Connecting to viewer at 127.0.0.1:9876
[... cut ...] BoardSocket.CAM_E: 4>, stream_enabled=False, name='Thermal', tof_align=None, is_used_as_stereo_align=False)] stereo=None ai_model=AiModelConfiguration(display_name='Yolo V6', path='yolov6nr3_coco_640x352', camera=<CameraBoardSocket.CAM_A: 0>) imu=ImuConfiguration(report_rate=100, batch_report_threshold=5)
No device selected, can't update pipeline!
Selecting device:  
[2024-11-18T17:02:36Z ERROR re_viewer::depthai::depthai] Error: No device selected, can't update pipeline!
Selecting device:  
```

MAVProxyUser, 2024-11-18:
I've got good news on OSX @Erol444 ! I've at least got somewhere to work now, even if my linux box isn't happy with it! 

[image]

```
MacBookAir:~ kfinisterre$ depthai-viewer 
Got exit status: Exited(0)
Log output: Requirement already satisfied: pip in /opt/homebrew/lib/python3.12/site-packages/depthai_viewer/venv-0.2.5/lib/python3.12/site-packages (24.2)
Collecting pip
  Using cached pip-24.3.1-py3-none-any.whl.metadata (3.7 kB)
Using cached pip-24.3.1-py3-none-any.whl (1.8 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 24.2
    Uninstalling pip-24.2:
      Successfully uninstalled pip-24.2
Successfully installed pip-24.3.1
Collecting depthai-sdk==1.14.0
  Downloading depthai_sdk-1.14.0-py3-none-any.whl.metadata (3.4 kB)
Collecting opencv-contrib-python>4 (from depthai-sdk==1.14.0)
  Downloading opencv_contrib_python-4.10.0.84-cp37-abi3-macosx_11_0_arm64.whl.metadata (20 kB)
Collecting blobconverter>=1.4.1 (from depthai-sdk==1.14.0)
  Downloading blobconverter-1.4.3-py3-none-any.whl.metadata (7.8 kB)
Collecting pytube>=12.1.0 (from depthai-sdk==1.14.0)
  Downloading pytube-15.0.0-py3-none-any.whl.metadata (5.0 kB)
Collecting depthai (from depthai-sdk==1.14.0)
  Downloading depthai-2.28.0.0-cp312-cp312-macosx_11_0_arm64.whl.metadata (8.7 kB)
Collecting PyTurboJPEG==1.6.4 (from depthai-sdk==1.14.0)
  Downloading PyTurboJPEG-1.6.4.tar.gz (11 kB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Collecting marshmallow==3.17.0 (from depthai-sdk==1.14.0)
  Downloading marshmallow-3.17.0-py3-none-any.whl.metadata (7.8 kB)
Collecting xmltodict (from depthai-sdk==1.14.0)
  Downloading xmltodict-0.14.2-py2.py3-none-any.whl.metadata (8.0 kB)
Collecting sentry-sdk==1.21.0 (from depthai-sdk==1.14.0)
  Downloading sentry_sdk-1.21.0-py2.py3-none-any.whl.metadata (8.6 kB)
Collecting depthai-pipeline-graph==0.0.5 (from depthai-sdk==1.14.0)
  Downloading depthai_pipeline_graph-0.0.5-py3-none-any.whl.metadata (8.8 kB)
Collecting ahrs==0.3.1 (from depthai-sdk==1.14.0)
  Downloading AHRS-0.3.1-py3-none-any.whl.metadata (13 kB)
Collecting numpy>=1.21 (from depthai-sdk==1.14.0)
  Using cached numpy-2.1.3-cp312-cp312-macosx_14_0_arm64.whl.metadata (62 kB)
Collecting Qt.py>=1.3.0 (from depthai-pipeline-graph==0.0.5->depthai-sdk==1.14.0)
  Downloading Qt.py-1.4.1-py2.py3-none-any.whl.metadata (29 kB)
Collecting packaging>=17.0 (from marshmallow==3.17.0->depthai-sdk==1.14.0)
  Downloading packaging-24.2-py3-none-any.whl.metadata (3.2 kB)
Collecting certifi (from sentry-sdk==1.21.0->depthai-sdk==1.14.0)
  Downloading certifi-2024.8.30-py3-none-any.whl.metadata (2.2 kB)
Collecting urllib3>=1.26.11 (from sentry-sdk==1.21.0->depthai-sdk==1.14.0)
  Downloading urllib3-2.2.3-py3-none-any.whl.metadata (6.5 kB)
Collecting requests (from blobconverter>=1.4.1->depthai-sdk==1.14.0)
  Downloading requests-2.32.3-py3-none-any.whl.metadata (4.6 kB)
Collecting PyYAML (from blobconverter>=1.4.1->depthai-sdk==1.14.0)
  Downloading PyYAML-6.0.2-cp312-cp312-macosx_11_0_arm64.whl.metadata (2.1 kB)
Collecting types-PySide2 (from Qt.py>=1.3.0->depthai-pipeline-graph==0.0.5->depthai-sdk==1.14.0)
  Downloading types_pyside2-5.15.2.1.7-py2.py3-none-any.whl.metadata (8.1 kB)
Collecting charset-normalizer<4,>=2 (from requests->blobconverter>=1.4.1->depthai-sdk==1.14.0)
  Downloading charset_normalizer-3.4.0-cp312-cp312-macosx_11_0_arm64.whl.metadata (34 kB)
Collecting idna<4,>=2.5 (from requests->blobconverter>=1.4.1->depthai-sdk==1.14.0)
  Downloading idna-3.10-py3-none-any.whl.metadata (10 kB)
Downloading depthai_sdk-1.14.0-py3-none-any.whl (230 kB)
Downloading AHRS-0.3.1-py3-none-any.whl (197 kB)
Downloading depthai_pipeline_graph-0.0.5-py3-none-any.whl (123 kB)
Downloading marshmallow-3.17.0-py3-none-any.whl (48 kB)
Downloading sentry_sdk-1.21.0-py2.py3-none-any.whl (199 kB)
[... 89 lines cut ...]
[2024-11-18 12:25:12] ERROR [depthai_sdk.logger.configure_syncing:69] Syncing requires at least 2 outputs! Skipping syncing.
Starting pipeline
[14442C104125EDCF00] [192.168.0.100] [9.249] [Camera(0)] [warning] PlgSrcTiny1c::Create: Capping fps from 30 to 25
Got message to send:  <depthai_viewer._backend.messages.WarningMessage object at 0x1348f63c0>
Sending message:  <depthai_viewer._backend.messages.WarningMessage object at 0x1348f63c0>
[2024-11-18T17:25:14Z WARN  re_viewer::depthai::depthai] Device is connected via PoE. This may cause performance issues.
----------------------------------------
[0s] System information
----------------------------------------
Ddr used / total - 9.76 / 333.26 MiB
Cmx used / total - 2.07 / 2.50 MiB
LeonCss heap used / total - 30.03 / 81.70 MiB
LeonMss heap used / total - 3.77 / 39.88 MiB
Chip temperature - average: 40.74, css: 42.58, mss: 40.28, upa: 40.04, dss: 40.04
Cpu usage - Leon CSS: 42.09 %, Leon MSS: 0.98 %
----------------------------------------
[9s] System information
----------------------------------------
Ddr used / total - 9.76 / 333.26 MiB
Cmx used / total - 2.07 / 2.50 MiB
LeonCss heap used / total - 30.42 / 81.70 MiB
LeonMss heap used / total - 3.92 / 39.88 MiB
Chip temperature - average: 42.81, css: 44.87, mss: 41.89, upa: 42.12, dss: 42.35
Cpu usage - Leon CSS: 22.65 %, Leon MSS: 50.67 %
```

MAVProxyUser, 2024-11-18:
@Erol444 I did indeed try again, to no avail from Linux, but it still responds fine from OSX (not simultaneously obviously).

Erol444, 2024-11-19:
Hi @MAVProxyUser ,my guess would be you have a problem with one of these networking issues (vpn, multiple interfaces, firewall..) on Linux, but not on Macos :

[image]

https://docs.luxonis.com/hardware/platform/deploy/poe-deployment-guide/#PoE%20deployment%20guide-Initial%20Connection-Debugging

Could you check this, please?

MAVProxyUser, 2024-11-19:
Please note it DID find the PoE device... it shows it in the list, the failure is when the driver attempts to connect. 

This is a brand new Ubuntu install. It is connected to the network exactly the same way as the MacOS machine, and happens to work just fine with my other PoE Luxonis camera. 

Here is my other Poe cam working... on the same network, nothing changed by swapping the camera out. 

```
kfinisterre@SeekOnJam0:~$ depthai-viewer 
[2024-11-19T16:14:54Z WARN  eframe::native::file_storage] Failed to parse RON: 4:14: Expected end of string
[2024-11-19T16:14:54Z INFO  winit::platform_impl::platform::x11::window] Guessed window scale factor: 1
vulkan: No DRI3 support detected - required for presentation
Note: you can probably enable DRI3 in your Xorg config
[2024-11-19T16:14:56Z WARN  egui_winit::clipboard] Failed to initialize arboard clipboard: Unknown error while interacting with the clipboard: X11 server connection timed out because it was unreachable
Backend started successfully.
[2024-11-19T16:14:59Z INFO  ewebsock::native_tungstenite] WebSocket handshake has been successfully completed
[2024-11-19T16:14:59Z INFO  re_viewer::depthai::ws] Websocket opened
Selecting device:  184430105124000900
[2024-11-19 10:15:00.996] [depthai] [warning] [184430105124000900] [192.168.0.120] Flashed bootloader version 0.0.15, less than 0.0.28 is susceptible to bootup/restart failure. Upgrading is advised, flashing main/factory (not user) bootloader. Available: 0.0.28
[2024-11-19 10:15:12.824] [depthai] [warning] Monitor thread (device: 184430105124000900 [192.168.0.120]) - ping was missed, closing the device connection
Connecting to viewer at 127.0.0.1:9876
Oak cam:  <depthai_sdk.oak_camera.OakCamera object at 0x746ce75969b0>
Default stereo pair:  None
Pipeline config:  auto=True cameras=[CameraConfiguration(fps=30, resolution=<CameraSensorResolution.THE_1080_P: 'THE_1080_P'>, kind=<CameraSensorType.COLOR: 0>, board_socket=<CameraBoardSocket.CAM_A: 0>, stream_enabled=True, name='Color', tof_align=None, is_used_as_stereo_align=False)] stereo=None ai_model=AiModelConfiguration(display_name='Yolo V6', path='yolov6nr3_coco_640x352', camera=<CameraBoardSocket.CAM_A: 0>) imu=ImuConfiguration(report_rate=100, batch_report_threshold=5)
Updating pipeline...
Creating auto pipeline config
Connected camera features:  [{socket: CAM_A, sensorName: IMX378, width: 4056, height: 3040, orientation: AUTO, supportedTypes: [COLOR], hasAutofocus: 1, hasAutofocusIC: 1, name: color}]
Usb speed:  UsbSpeed.UNKNOWN
Connected to a PoE device, camera streams will be JPEG encoded...
Creating camera:  fps=30 resolution=<CameraSensorResolution.THE_1080_P: 'THE_1080_P'> kind=<CameraSensorType.COLOR: 0> board_socket=<CameraBoardSocket.CAM_A: 0> stream_enabled=True name='color' tof_align=None is_used_as_stereo_align=False
Connected cam doesn't have IMU, skipping IMU creation...
Starting pipeline
Got message to send:  <depthai_viewer._backend.messages.WarningMessage object at 0x746ce75e19c0>
Sending message:  <depthai_viewer._backend.messages.WarningMessage object at 0x746ce75e19c0>
----------------------------------------
[1s] System information
----------------------------------------
Ddr used / total - 96.90 / 333.26 MiB
Cmx used / total - 2.49 / 2.50 MiB
LeonCss heap used / total - 41.00 / 81.70 MiB
LeonMss heap used / total - 4.36 / 39.88 MiB
Chip temperature - average: 35.82, css: 37.71, mss: 34.40, upa: 35.11, dss: 36.06
Cpu usage - Leon CSS: 36.94 %, Leon MSS: 0.05 %
[2024-11-19T16:15:26Z WARN  re_viewer::depthai::depthai] Device is connected via PoE. This may cause performance issues.
----------------------------------------
[10s] System information
----------------------------------------
Ddr used / total - 109.08 / 333.26 MiB
Cmx used / total - 2.49 / 2.50 MiB
LeonCss heap used / total - 41.89 / 81.70 MiB
LeonMss heap used / total - 5.21 / 39.88 MiB
Chip temperature - average: 37.47, css: 39.35, mss: 36.77, upa: 37.48, dss: 36.30
Cpu usage - Leon CSS: 19.58 %, Leon MSS: 1.49 %
```

And oddly enough. Immediately after that camera worked, now the Thermal decided it is gonna cooperate. (after two days of refusing to)

[image]

Watching the device selection list the two IP's keep rotating in the list (until I select one). I wonder if the PoE discovery code is just a bit flakey, and maybe attempts too many times to discover the device after it already knows where it is? Perhaps triggering some sort of race condition? 

Either way. I did get the camera to view finally! 

I did note specifically "ping was missed, closing the device connection", could it be as simple as too aggressive of a keep alive check? Surely missing more than ONE ping would be the way to go before closing the connection out? 

Thanks for the troubleshooting help!