luxonis/depthai #1222: [BUG] Windows Defender detects a virus in DepthAI Demo 3.6.1 - Trojan:Win64/Bastdoor.MKV!MTB

McMornan, 2024-12-08:
Title says it all ... tried to install depthai demo 3.6.1 here from Github on win11 and defender found the trojan above. File: AppData\Local\Programs\DepthAI\venv\Lib\site-packages\depthai_viewer_bindings\depthai_viewer_bindings.pyd

marsfan, 2024-12-23:
Hey. Not a Luxonis Employee, but someone else that ran into the same problem, and has done some investigating. Its an issue with the viewer specifically, and they are working on it. The relevant bug is here: https://github.com/luxonis/depthai-viewer/issues/78

McMornan, 2024-12-23:
Thanks, will check it out ! For the time being I have disabled the check to run the demo but feels unsafe ...

Erol444, 2025-01-17:
Seems like this has been fixed on 0.28 by updating toolchain:
https://github.com/luxonis/depthai-viewer/commit/75278ea3b36aeafeb00637bf433c222d1924e3c9

closed 2025-01-17 by Erol444