luxonis/depthai #1230: Trying to connect to POE cameras remotely through 4G router

GuillaumeHupi, 2025-03-07:
Hello, 

We are using OAK-1 PoE cameras. They are deployed on site and connected via a 4G router. This all works, but if we need to do a software update, for now we go on site, connect to the router to be on the same network as the camera and run the python script to flash a firmware in standalone mode.

We will have to deploy the solution on sites quite far away and we are looking for a way to be able to make updates remotely. Our telephone operator provides us with VPN access and the IP of our SIM card once connected to them in VPN in the format 10.X.X.X. Thus, we can for example connect to the router remotely using this IP.

On our router we can also configure a port redirection, So for example, if we connect a computer to the router and if we configure a port redirection for port 22 on the computer's IP. Then, we can remotely access this computer in ssh using the SIM IP. For example if the IP of my SIM is 10.10.10.25. If I connect to the operator's VPN and I run a command "ssh root@10.10.10.25", I am well connected in ssh to my computer connected by ethernet to the 4G router.

So we were able to validate that the port redirection works well. We then wanted to see if it was possible to access the POE camera with the same method.

In doubt, on the router we configured port forwarding for all ports (0-65535) to the camera IP. Because we analyzed the requests made and several ports are used (at least 90 for ping and 11490 for firmware flash).

And then, we just tried to launch the command

```
dai.DeviceBootloader.getFirstAvailableDevice()
```

But it could not find a device. We thought it was related to the fact that the camera IP is not on the same network mask as the computer on which we are running the program (camera in 10.X.X.X, while the computer is in 192.168.1.X). So we wanted to specify the IP through this command:

```
bootloader = depthai.DeviceBootloader(ip_address, False)
```

But remotely, it is impossible to detect the camera:

```
bootloader = depthai.DeviceBootloader(ip_address, False)
RuntimeError: Specified device not found
```

But we are able to ping the camera on the 10.X.X.X IP so I don't uderstand why the python script is not able to find it. 

Do you have any idea what causes this behavior? For example, does the python library encounter a problem when we try to use an IP that is on a different address range than the computer on which the script is executed? Or is there a particular setting that we should use on our router?

themarpe, 2025-03-10:
@GuillaumeHupi  (CC: @jakaskerl )

You'll need to check if UDP 11491 is also passed through as that is required for both "device discovery" as well as when specific IP is set (for device state check)

GuillaumeHupi, 2025-03-18:
@themarpe , 

Sorry for my late response. 
We made different tests. I don't know why; but when forwarding all ports, it wasn't working, but when forwarding only 11491 and 11490 ports on our router, we are able to detect the device remotely through the VPN. 

But we are not able to flash a new script remotely. I found this bug https://github.com/luxonis/depthai-core/issues/794 and we are in exaclty the same scenario and with the same result. 

I am not sure I understood what was said in the discussion. From what I understood, the problem is caused by the bootloader that is not reseted when we flash firmware to the camera. And so it is triggered before the flash is done and the camera reboot which stops the flash process before the end and causes an error. And this is due to 4G latency which make the process longer than via direct ethernet. Am I wrong ?

If I'm right, a trick to make it work would be to extend the watchdog or to totally disable it. I tried to flash my script while specifying the the watchdog and bootup delays like this : 
`DEPTHAI_WATCHDOG_INITIAL_DELAY=60000 DEPTHAI_BOOTUP_TIMEOUT=60000 python oak_detection_lifeline.py` 

Or to totally disable the watchdog like this : 
`DEPTHAI_WATCHDOG=0 python oak_detection_lifeline.py`

But I still got the same problem. In the github discussion, it was mentioned to also make some modifications to do inside the C library, but I don't know where to find this library when I use the depthai-python library inside a script. 

So I'm a bit lost. Could you explain to me the detailed procedure on how to disable the watchdog, in order to make tests and be sure that this is the watchdog that causes us this problem ? 

Thanks

Guillaume

themarpe, 2025-03-18:
CC: @moratom if you could help @GuillaumeHupi with a artifacts of a built python package against latest + added this patch on top (from the other issue's thread): https://github.com/luxonis/depthai-core/compare/main...mrmorawski:depthai-core:no_bootloader_watchdog and or we bring it in as DEPTHAI_BOOTLOADER_WATCHDOG env var & bake it in. (note, naive impl. won't work fully, as it doesn't change device's WD, but BL has an internal bump per any "comms received" so its should work for most cases)

moratom, 2025-03-20:
@GuillaumeHupi  I've added internal tracking on this and will update the case once we have the artifact build ready.

GuillaumeHupi, 2025-03-24:
We have a blocking point in our system, do you have any idea of a delivery date for this please ?

moratom, 2025-03-24:
Hi @GuillaumeHupi - wheels are building here https://github.com/luxonis/depthai-python/actions/runs/14039334687.

After they are built for your needed platform, you will be able to install with
```
python3 -m pip install -U --prefer-binary --extra-index-url https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local "depthai==2.30.0.0.dev0+0625251d8460388fc2aa5f81546b10499a6a3c92"
```

to "turn off" the watchdog, set `DEPTHAI_WATCHDOG=0` and it will disable the bootloader watchdog as well.

Hopefully this unblocks you, otherwise let us know.

GuillaumeHupi, 2025-03-25:
Hi @moratom , 

Thank you very much, I was able to install it with the command you sent. 

Then I tried to upload the firmware with my script with watchdog disabled by using a command like this : 
`DEPTHAI_WATCHDOG=0 DEPTHAI_LEVEL=debug python myscript.py`
Am I doing well ? 

If so, I have a strange behavior. Sometimes, the upload works fine, sometimes no. And I have a similar behavior with the DEPTHAI_WATCHDOG argument to 0 or 1. This does not seem to have a big impact (it seems to work a little bit more often when watchdog is disabled, but I'm not sure). 

When I have errors, it is always related to X_LINK_TIMEOUT : 

```
DEPTHAI_WATCHDOG=0 DEPTHAI_LEVEL=debug python oak_detection_lifeline.py
[2025-03-25 15:49:51.937] [depthai] [debug] Python bindings - version: 2.30.0.0.dev0+0625251d8460388fc2aa5f81546b10499a6a3c92 from 2025-03-23 11:19:48 +0100 build: 2025-03-24 16:39:57 +0000
[2025-03-25 15:49:51.937] [depthai] [debug] Library information - version: 2.30.0, commit: e0f6b52d048ef7ceac2ecd44deb2e161dbb97699 from 2025-03-18 12:44:00 +0100, build: 2025-03-24 16:39:56 +0000, libusb enabled: true
[2025-03-25 15:49:51.943] [depthai] [debug] Initialize - finished
[2025-03-25 15:49:52.010] [depthai] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.28.tar.xz' open: 1ms, archive read: 68ms
[2025-03-25 15:49:52.213] [depthai] [debug] Found an actual device by given DeviceInfo: DeviceInfo(name=10.44.199.197, mxid=14442C1021A6ADCF00, X_LINK_BOOTED_NON_EXCLUSIVE, X_LINK_TCP_IP, X_LINK_MYRIAD_X, X_LINK_SUCCESS)
[2025-03-25 15:49:52.311] [depthai] [debug] Resources - Archive 'depthai-device-fwp-a62b2ccb0bc493c2fb41694cb81c08887be24c52.tar.xz' open: 1ms, archive read: 371ms
[2025-03-25 15:49:59.538] [depthai] [debug] Searching for booted device: DeviceInfo(name=10.44.199.197, mxid=14442C1021A6ADCF00, X_LINK_BOOTLOADER, X_LINK_TCP_IP, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[2025-03-25 15:50:07.340] [depthai] [warning] Monitor thread (device: 14442C1021A6ADCF00 [10.44.199.197]) - ping was missed, closing the device connection
[2025-03-25 15:50:09.343] [depthai] [debug] XLinkResetRemoteTimeout returned: X_LINK_TIMEOUT
[2025-03-25 15:50:09.343] [depthai] [debug] XLinkResetRemote of linkId: (0)
[2025-03-25 15:50:09.850] [depthai] [debug] Searching for booted device: DeviceInfo(name=10.44.199.197, mxid=14442C1021A6ADCF00, X_LINK_BOOTLOADER, X_LINK_TCP_IP, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[2025-03-25 15:50:23.183] [depthai] [debug] Connected bootloader version 0.0.28
[2025-03-25 15:50:23.299] [depthai] [debug] DepthAI Application Package
[2025-03-25 15:50:23.300] [depthai] [debug] __firmware, 27489219B, 512, 2795360291, 0, 3
[2025-03-25 15:50:23.300] [depthai] [debug] pipeline, 1586B, 28311552, 3285115098, 0, 0
[2025-03-25 15:50:23.300] [depthai] [debug] assets, 56B, 28377088, 2456244843, 0, 0
[2025-03-25 15:50:23.300] [depthai] [debug] asset_storage, 5994328B, 28442624, 4175481663, 0, 0
[2025-03-25 15:50:23.300] [depthai] [debug] __fw_version, 40B, 34471936, 285283003, 0, 0
[2025-03-25 15:50:23.300] [depthai] [debug] app_name, 0B, 34537472, 5381, 0, 0
[2025-03-25 15:50:29.928] [depthai] [warning] Monitor thread (device: 14442C1021A6ADCF00 [10.44.199.197]) - ping was missed, closing the device connection
[2025-03-25 15:50:31.933] [depthai] [debug] XLinkResetRemoteTimeout returned: X_LINK_TIMEOUT
[2025-03-25 15:50:31.933] [depthai] [debug] XLinkResetRemote of linkId: (14)
Traceback (most recent call last):
  File "/Users/guillaumecostes/Projets/adt_smartcounting/oak_detection_lifeline.py", line 230, in <module>
    bootloader.flash(progress, pipeline)
    ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^
depthai.XLinkWriteError: Couldn't write data to stream: '__bootloader' (X_LINK_ERROR)
[2025-03-25 15:50:31.966] [depthai] [debug] DeviceBootloader about to be closed...
[2025-03-25 15:50:32.435] [depthai] [debug] DeviceBootloader closed, 469
```

So I am not sure if the watchdog is well disabled when I add the argument DEPTHAI_WATCHDOG=0.

moratom, 2025-03-26:
I see, this is unfortunally expected as per
>note, naive impl. won't work fully, as it doesn't change device's WD, but BL has an internal bump per any "comms received" so its should work for most cases

since we don't actually disable the watchdog on the device, only on the host side of DepthAI.
Can you work around this by doing multiple retries?